t6x/reaver-wps-fork-t6x #199: in json mode, wash never prints APs which have WPS on, but no probe was received

rofl0r, 2018-01-31:
this is because we wait for the probe request so we can print the entire info in one line.

possible solutions:

- record information in beacon of those APs (maybe by saving an entire beacon packet) and on EOF (if no more records are in pcap) print the data of the missing APs (possibly by calling the routine again with the saved beacon)

- we could print a line without extended WPS info as soon as a beacon is received, and later when we have the full data print another line with the full info

kcdtv, 2018-02-07:
Hey! sorry for not answering before, i saw the issue and did not realy knew what to think. 
I am still in this mood... For me it is not an issue as it is... what is expected. 
I use fast scan without -j and then if i focus a target I know where it is.
Anyway, the first option you propose sounds horrible, the second one would be better in my opinion.

rofl0r, 2018-02-07:
could you elaborate on why you think the first option sounds horrible ?

kcdtv, 2018-02-08:
Saving data and retro check them sounds "horrible" (ugly) to me comparing to doing the job on the fly without storing any beacon or traffic capture. The second proposal seems much more "elegant". 
And efficient-simpler-more straightforward as a process.