t6x/reaver-wps-fork-t6x #201: wash does not work with tee and creates mon interfaces at every start

wpa2, 2018-02-22:
root@porteus:/# airmon-ng 

PHY	Interface	Driver		Chipset

phy0	wlan0		brcmsmac	Broadcom on bcma bus, information limited
phy1	wlan1mon	rt2800usb	Ralink Technology, Corp. RT5370

root@porteus:/# wash -i wlan1mon
BSSID               Ch  dBm  WPS  Lck  Vendor    ESSID
--------------------------------------------------------------------------------
4C:09:B4:xx:xx:xx    6  -79  1.0  No   AtherosC  UKrtelecomxxxxxx
D4:6E:0E:xx:xx:xx   11  -71  2.0  No   RalinkTe  Myxxxx
C8:3A:35:xx:xx:xx    7  -85  1.0  No   Broadcom  Tendaxxxxxx
^C
root@porteus:/# wash -i wlan1mon | tee 123.txt
^C
root@porteus:/# ls -l 123.txt 
-rw-r--r-- 1 root root 0 фев 22 14:49 123.txt
root@porteus:/# airmon-ng 

PHY	Interface	Driver		Chipset

phy1	mon0		rt2800usb	Ralink Technology, Corp. RT5370
phy1	mon1		rt2800usb	Ralink Technology, Corp. RT5370
phy0	wlan0		brcmsmac	Broadcom on bcma bus, information limited
phy1	wlan1mon	rt2800usb	Ralink Technology, Corp. RT5370

root@porteus:/# wash

Wash v1.6.4 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner
...

https://github.com/t6x/reaver-wps-fork-t6x/
Checked out revision 602.

rofl0r, 2018-02-22:
revision 602 ? this is not SVN. sha1 hash or it didn't happen.

kcdtv, 2018-02-23:
Same problem here (tested in Kali Linux) with latest commit:
```
kcdtv@kalimuX0:~$ time sudo wash -i wlan0mon | tee /tmp/shit
^C

real	0m8,780s
user	0m0,004s
sys	0m0,092s
kcdtv@kalimuX0:~$ cat /tmp/shit
kcdtv@kalimuX0:~$ 
```
Nothing in the shell, nothing in the file. 
There is not such an issue with version **1.6.4** stable, everything goes fine.

rofl0r, 2018-02-23:
the tee issue should be fixed. @wpa2 please open a new issue for the monitor interface issue you experience, but with more details.

closed 2018-02-23 by rofl0r

wpa2, 2018-02-23:
Ok. I will test some more with various adapters and other programs working with wifi, and I will write when there is more information.
Thanks!

rofl0r, 2018-02-23:
great. @kcdtv i see you added that screenshot, did you test if my commit fixed the issue ?

wpa2, 2018-02-24:
version 603 now works with tee.
Thanks!

rofl0r, 2018-02-24:
could you please stop refering to git commit in svn lingo, thanks

kcdtv, 2018-02-24:
Hey!  I've tested the "revision 603" :smiley_cat: and it solved the problem observed with piped comand. We can keep this  issue close.

commit 48b13810 (2018-02-23): wash: disable stdio buffering

closes #201