t6x/reaver-wps-fork-t6x #202: monN interfaces from wash.

wpa2, 2018-02-24:
When you run reaver and wash, create the interface monN (N the minimum unused digit is 0, if 0 is busy then 1 etc). And monN is created even if the interface specified by the -i option is in the monitor mode.
But after stopping, reaver deletes the interface created, and wash does not delete it. As a result, after several launches of wash, we have several interfaces - mon0, mon1, mon2, etc.

P.S. As I understand now reaver and wash it is the same file. Maybe you can do a scan in reaver and remove an extra copy of the file named wash?

rofl0r, 2018-02-24:
> When you run reaver and wash, create the interface monN (N the minimum unused digit is 0, if 0 is busy then 1 etc). 

did you put your interface manually into monitor mode before starting reaver/wash ?

also, on which system does this happen ? please specify OS version and architecture( `eg. Ubuntu 16.04 x86_64`)

> But after stopping, reaver deletes the interface created, and wash does not delete it

interesting.

> P.S. As I understand now reaver and wash it is the same file. Maybe you can do a scan in reaver and remove an extra copy of the file named wash?

what extra copy ? `wash` is a symlink to `reaver` and the reaver binary decides upon which functionality to start by looking at the name of how it was started.

wpa2, 2018-02-24:
> did you put your interface manually into monitor mode before starting reaver/wash ?

Yes.
If the interface is not switched to monitor mode, monN is still created, but the scan does not go. Just appears the title of the table and nothing else happens.

> also, on which system does this happen ? please specify OS version and architecture( eg. Ubuntu 16.04 x86_64)

```
# uname -a
Linux porteus 3.19.8-porteus #2 SMP PREEMPT Mon Jul 6 16:29:21 MSK 2015 x86_64 Intel(R) Core(TM) i7-2630QM CPU @ 2.00GHz GenuineIntel GNU/Linux
```
> interesting.

ran and stopped the wash three times:
```
# airmon-ng 

PHY	Interface	Driver		Chipset

phy1	mon0		rt2800usb	Ralink Technology, Corp. RT5370
phy1	mon1		rt2800usb	Ralink Technology, Corp. RT5370
phy1	mon2		rt2800usb	Ralink Technology, Corp. RT5370
phy0	wlan0		brcmsmac	Broadcom on bcma bus, information limited
phy1	wlan1		rt2800usb	Ralink Technology, Corp. RT5370

# iwconfig 
eth0      no wireless extensions.

mon0      IEEE 802.11bgn  Mode:Monitor  Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
wlan0     IEEE 802.11bgn  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=0 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          
mon1      IEEE 802.11bgn  Mode:Monitor  Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
mon2      IEEE 802.11bgn  Mode:Monitor  Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
lo        no wireless extensions.

wlan1     IEEE 802.11bgn  ESSID:off/any  
          Mode:Managed  Frequency:2.417 GHz  Access Point: Not-Associated   
          Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off

```
> what extra copy ? wash is a symlink to reaver and the reaver binary decides upon which functionality to start by looking at the name of how it was started.

It's my script create a second copy instead of a symlink. Sorry.

rofl0r, 2018-02-24:
that's a pretty old kernel you got there. probably a driver bug, or a similarlly ancient, buggy pcap lib.
as for why the interface gets removed with reaver but not with wash:
maybe set a breakpoint on `pcap_close` and see if it gets called.

ianox, 2018-04-15:
I can actually reproduce this behaviour using manjaro Linux 4.9.94-1-MANJARO x86_64 17.1.8 Hakoila. The driver for my inbuilt Intel wifi chip is iwlwifi .
Again, everytime wash is run and stopped, a new monitor device monX remains unclosed.
Let me know if I can be of further assistance.

wifislax-ng, 2018-06-02:
Hi , same here ..

kernel 4.14.47
iw 4.3
libpcap-1.8.1

I put wlan0 on monitor with airmon-ng (aircrack-1.2) , i get wlan0mon
run

wash -i wlan0mon

and i see now using iwconfig , new interface enerated by wash ... mon0 + wlan0mon

rofl0r, 2018-06-02:
@wifislax-ng : which chipset/driver ?

wifislax-ng, 2018-06-02:
rofl0r , this issue, is for all chips i think , i test and get monX interfaces ,with this chipsets ..all tested, all afected.

Ralink Technology, Corp. RT2870/RT3070
Realtek Semiconductor Co., Ltd. RTL8723B
Atheros Communications, Inc. AR9271 802.11n

wifislax-ng, 2018-06-02:
Hi again , afetr some test ...

reaver 1.6.3 is the latest working as expected ...  1.6.3 no generate a monX interfaces

1.6.4 & 1.6.5 , are affected.

wifislax-ng, 2018-06-03:
Hi again , and sorry for this , i download and test , commit by commit , since 1.6.3 , and the first commit and the culprit ,is this one

https://github.com/t6x/reaver-wps-fork-t6x/commit/bdd420d7fe4114f6e5da46654b8c95cc22fd042b

This is the first commit making monX interfaces after use wash.

kcdtv, 2018-06-03:
well play wifislax-ng! Apple is evil! :smiley_cat:

rofl0r, 2018-06-03:
i already suspected it could have to do with that commit... but i don't really get why setting the rfmon flag would create a new device... this sounds like a bug in either pcap or the kernel to me; especially considering that it works for almost everybody as expected. i think i'll have a 4.14 kernel ready soonish so i can test it myself, in the worst case we'll have to have make something like 
```
#ifdef APPLE
rfmon_stuff();
#endif
```
but i'd rather get to the gist of this.

rofl0r, 2018-06-03:
hmm, it might also have to do with how airmon-ng sets up and renames the monitor interface. i usually use macgeiger to put my adapter into monitor mode, which uses much simpler mechanics to do (it basically just switches monitor mode on).

kcdtv, 2018-06-04:
**Kernel**: I used kernel 4.14 in debian and derivative OS such as Ubuntu and Kali and I never had such an issue... I doubt that you will be able to reproduce the error just by upgrading your kernel.   
**Libpcap**: i have the same version than wifislax-ng and no problems. 
If it is one of the two things, kernel or libpcap, I would say that the kernel is probably more modified from a disitrbution to another than libpcap. 
**Airmon-ng**: I always use it to activate the monitor mode and never got this error. 
Maybe wifislax-ng or ghost or wpa2 could try to activate mode monitor with a simple `iwconfig` command and see if the issue happens again. Saying so i don't really see how the two things would be linked... if there is a single interface in monitor mode before using reaver or wash; why would airmon-ng have something to do with the fact that stopping reaver or wash generate another monitor mode interface? I don't see the relation here.
saying so, even if that would be a problem with airmon-ng, kernel variation or libpcap, I reommend to adapt in order to bring support to slackware and arch linux based system. Can you even use reaver in a macOS system? Great if you can compile but, as far as i know, there are no drivers for injecting, so it is what i would even call a lost of time. And the focus here are GNU-Linux distributions, not a non free system designed for expensive hardware that not everybody can afford. Above all, a mac user can always install a free GNU-Linux distribution using for example refind (or redit, i don't remember which one is deprecated) which is free too. Maybe wifislax-ng or ghsot or wpa2 can try to compile again reaver modifying the rfmon line and tell us what happen too... That would be helpful too.

rofl0r, 2018-06-04:
> why would airmon-ng have something to do with the fact that stopping reaver or wash generate another monitor mode interface?

well, my guess is that the renaming of the interface causes a sort of "shadow interface" to still exist with the old interface name linked to the interface, and somehow when using the rfmon activation of libpcap that gets "awoken" too.
it's also imaginable that @wifislax-ng runs `airmon-ng start wlan1` and then uses `wash -i wlan1` instead of `wlan1mon`, which then causes the "shadow interface" to re-surface.

> And the focus here are GNU-Linux distributions

i agree, so maybe we should just add the above mentioned ifdef and be done with this.

kcdtv, 2018-06-04:
> i agree, so maybe we should just add the above mentioned ifdef and be done with this.

:+1:

wifislax-ng, 2018-06-05:
rofl0r , when use airmon-ng wlanX dissapears , and goes renamed to wlanXmon , its impossible go then with wlanX cause after airmon-ng that non exists.

If you do , airmon-ng start wlan0 , then ..wlan0 dissapears , and have only wlan0mon

wifislax-ng, 2018-06-06:
Hi again , i finally  , GOT IT!!!

Problem is in libpcap, when building with libnl support , then have the ability to create interfaces on monitor ... 

I rebuild my libpcap-1.8.1 , adding

--without-libnl

and no more monX interfaces.

Can test if libnl are linked, with ldd

ldd /usr/lib/libpcap.so

I dont now , how fix this... because libpcap are provided by distros ... and some build with libnl , others without ..etc etc.

kcdtv, 2018-06-06:
Very nice job! That is tricky....

wifislax-ng, 2018-06-06:
Easy way to check if libpcap , are linked to libnl ...

pcap-config --static --libs

In my case results is yes .. libnl are linked , in archlinux same results ..

-L/usr/lib64 -lpcap  **-lnl-genl-3 -lnl-3**  -ldbus-1

rofl0r, 2018-06-08:
interesting. might it actually be dbus, which is pulled in by libnl, which is responsible for this odd behaviour ? that would be my first suspicion, since dbus is the product of the systemd-maniac lennart poettering, who seems to be a fountain of horrible ideas.

kcdtv, 2018-06-08:
archlinux uses systemd but slackware sticked to init (correct me if i am wrong wifislax-ng)

rofl0r, 2018-06-08:
libnl certainly can be compiled --without-dbus; and i could imagine that this would fix the issue..

wifislax-ng, 2018-06-08:
I explain all you need to know.

First commit when start the issue , and the reason ..

libpcap  ---> linked to libnl ...

./configure --prefix=/usr --without-libnl  to rebuild libpcap without libnl support .. but this is not a solution ..

People no go to rebuild "custom" package from libpcap , cause later probably break other "distro release" packages linked to the original libpcap.

slackware no have systemd , its remain ti init and eudev , but the issue still present in other distros with systemd , like archlinux...or majaro.

You ever think problem is in other part ... but no , reaver-1.6.3 works perfectly , ...problem is in reaver code, in "some enviroments" . 

I test some new an d old kernels cayuse you start saying ... later i start commit by commit and encounter the exactly commit causing this ... later i discover exactly reason .. and you think arround systemd , or libdbus ....   

You have here all that you need to know, patch or not patch ... you decide.

wifislax-ng, 2018-06-08:
This is a very long history ... all distros have libcap with libnl support except debian and derivatives, cause debian developers choose no link.

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=651783

rofl0r, 2018-06-09:
> slackware no have systemd

i never mentioned systemd related to this issue, only related to the author of DBUS.

> i discover exactly reason .. and you think arround systemd , or libdbus
> problem is from reaver code , accept this.

i'm well aware which change in reaver code causes the issue to pop up, but i'm trying to find the *root* of the cause, not just the symptom...

and since you were doing tests with pcap with and without libnl, i had the impression you were willing to do some tests so we can **really** get to the bottom of this.

so, could you *please* be so kind as to compile libnl without dbus, and then tell us if that removes the issue ?

kcdtv, 2018-06-09:
Let's try to focus on fully understanding the issue with a positive attitude, very good job have been done so far, i am pretty sure we are very close.  :smiley_cat:

DJDan, 2018-06-10:
@kcdtv Reaver/Wash/Monitor Mode/Injecting indeed works on a MAC. I am using a Macbook Pro Mid 2010 17" Laptop with macOS SIerra 10.12.6. Both Reaver 1.6.5 and Wash works amazingly so far.. :)
@rofl0r If you do happen to change Mac specific code please private message me on here and let me know so I can retest and make sure it is still working.

kcdtv, 2018-06-10:
Hi DJDan! Thanks for correcting me and I am glad to learn that.  I see that the wifi chipset is a bcm43xx. Just by curiosity: Did you have to install extra/non-official dirvers or firmware or does it works "out of the box"? We will for sure have it presents in mind and won't drop support (that was not the plan in any case).

rofl0r, 2018-06-12:
> @rofl0r , you no understand nothing.

you're growing a big mouth, dude.

> my libnl , no have dbus

so where's this dbus reference coming from ? go figure.

> pcap-config --static --libs
> -L/usr/lib64 -lpcap -lnl-genl-3 -lnl-3 -ldbus-1

> The only game here is ...

wrong. i happen to create my [own linux distribution from scratch (and from source)](http://sabotage.tech) and in my case libpcap is linked against libnl-tiny.

> luke , use the ldd force ...

ldd is a tool for noobs. i use readelf.

kcdtv, 2018-06-12:
Please guys:  :peace_symbol: 
@rofl0r: Keep in mind that **wifislax-ng** does not speak very well English (neither do I, it is not a critic)... Do not take personally some clumsy sentences that might sounds a bit aggressive. It is just a lack of vocabulary
@wifislax-ng: rofl0r **nunca habla por hablar**... Conoce muuuy bien el lenguaje C y el sistema Linux. No se trata de tener razón o no: Se trata de entender un fallo y no quedarse a la superficie de las cosas. Y aprender todos juntos con un espíritu constructivo...  
Si te ha pedido de compilar las libnl sin dbus es por algo. Te toma menos tiempo que escribir este mensaje y es mucho menos tiempo que bajar y instalar y configurar arch linux para hacer pruebas. Que tampoco sería muy instructivo porque son sistemas que  vienen ya muy "tuneados". Así que, por favor, seamos cortés y amables... Aplicando a nosotros mismos la filosofía del software libre en lugar de entrar en discusiones y enfadarse para cosas sin sentido. :smile_cat:

wifislax-ng, 2018-06-17:
Second time ... mu libnl , NOT LINKED WITH DBUS.

pcap-config --static --libs

Its from libpcap , understand this?

My libnl never linked to dbus , when you execute

pcap-config --static --libs

you have libs linked to libpcap.

And as i say , ldd arround libnl ...says no dbus linked.

where see you libnl linked to dbus ?
ldd /usr/lib64/libnl.so
        linux-vdso.so.1 (0x00007ffde4a88000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f4220d59000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f4220b3c000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f4220773000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f42212b3000)

Read slow , and understand first ... 

pcap-config is a script tool from LIBPCAP , and say what libraries are linked to --> libpcap ...or if you understand more easy ...

ldd /usr/lib64/libpcap.so
        linux-vdso.so.1 (0x00007fff0e3da000)
        libnl-genl-3.so.200 => /usr/lib64/libnl-genl-3.so.200 (0x00007fbb8f189000)
        libnl-3.so.200 => /usr/lib64/libnl-3.so.200 (0x00007fbb8ef69000)
        libdbus-1.so.3 => /usr/lib64/libdbus-1.so.3 (0x00007fbb8ed1a000)
        libc.so.6 => /lib64/libc.so.6 (0x00007fbb8e951000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fbb8e734000)
        libm.so.6 => /lib64/libm.so.6 (0x00007fbb8e42b000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fbb8f5df000)

As you can see ...libdbus appears linked to libpcap , and never to libnl.

The problem ... is libpcap linked to libnl  ...if rebuild my libpcap withour libnl support  (like debian distros) , then no issues.

But say users , go to rebuild your libpcap , is not the way.

rofl0r, 2018-06-17:
ok. so then it is related to libnl, not dbus.

i guess we should study the code of libnl3 and look for rfmon and why it tries to create a new monitor interface.
note that this doesn't happen with libnl-tiny, which i use.

wifislax-ng, 2018-06-17:
@glezo1

No , .. read the thread , archlinux , manjaro and others are affected ... not good idea overlap stock distro packages from "user custom packages".

@rofl0r , you can have libpcap + libnl ,  cause when developers build libpcap can pass the argument --without-libnl to the libpcap compilation , debian do this exactly ,diable when create the libpcap package.

Other distros like archlinux , want explictly enable this .. putting libnl as "make depend" , caus do  --with-libnl  in compilation time.

Please see, archlinux libpcap build , ... force libnl to enhanced libpcap functions.
https://git.archlinux.org/svntogit/packages.git/tree/trunk/PKGBUILD?h=packages/libpcap

THe questions here , is what extra capabilities do libnl for libpcap , affecting to the monitor mode check code in wash.

wifislax-ng, 2018-06-17:
See archlinux forcing same as slackware ...
depends=('glibc' 'libnl' 'sh' 'libusbx' 'dbus')

Problem arrives in a systems with libpcap and libnl capabilities enabled.

rofl0r, 2018-06-17:
i pushed a fix, please try @wifislax-ng, thanks!

wifislax-ng, 2018-06-18:
No lucky , i see new monX interfaces when wash start ... at start scan make mon interfaces ..
wash -i wlan0mon   , and after 2 test

iwconfig
eth0      no wireless extensions.

mon0      IEEE 802.11  Mode:Monitor  Frequency:2.437 GHz  Tx-Power=20 dBm   
          Retry short limit:7   RTS thr=2347 B   Fragment thr:off
          Power Management:on
          
lo        no wireless extensions.

wlan0mon  IEEE 802.11  Mode:Monitor  Frequency:2.437 GHz  Tx-Power=20 dBm   
          Retry short limit:7   RTS thr=2347 B   Fragment thr:off
          Power Management:on
          
mon1      IEEE 802.11  Mode:Monitor  Frequency:2.437 GHz  Tx-Power=20 dBm   
          Retry short limit:7   RTS thr=2347 B   Fragment thr:off
          Power Management:on

wifislax-ng, 2018-06-18:
I see you comment patch is for interrump process , but interfaces appears in a same time process start.

rofl0r, 2018-06-18:
well the original issue reported was that the monX interfaces **stay** after wash is closed - that's what this patch tries to fix. it's not a problem when an interface shows up as long as wash is running.

kcdtv, 2018-06-18:
Issue is solved (tested in balckarch), well played! To be clear: An extra interface will shows up while you run wash or reaver but it disappears as soon as you stop the running process, leaving the system as it was before running wash.

rofl0r, 2018-06-19:
yeah, this was the originally reported problem. it's arguably a bit nasty that pcap creates new devices behind one's back, so we might still want to disable the corresponding code in reaver.
but maybe we should open a dialogue with pcap devs and ask whether this can be implemented without creating a new interface.

wifislax-ng, 2018-06-19:
pcap_set_rfmon(handle, 1); --> pcap_set_rfmon(handle, 0);

rofl0r, 2018-06-19:
yes, i'm aware of that, but that change would break mac support.

wifislax-ng, 2018-06-19:
no way to do a check , and if machine is mac , then run the pcap monitor check ?

wifislax-ng, 2018-06-27:
Patch ,not work good .........
wash running in one konsole
open other konsole ...  killall wash  ... monX not deleted.

If you think this is absurd , think arround scripts, running some automated works ... when try to kill wash , he dies , but monX not.  :=)

rofl0r, 2018-06-27:
`killall wash` - why would you do that ? lazyness ?

kcdtv, 2018-06-28:
Is it different if you use `killall -s SIGINT wash` ?

wifislax-ng, 2018-06-28:
@rofl0r 

Some automated scrips , ask the user for time scan from wash ..

example , scan wash 30 seconds .. after 30 seconds the script killall wash to esto the process , then , cause the signal is from script and not directly in a konsole, then wash stop but monX not.

Probably other tools , like inflator , do the same after time scan ... i go test and return.

wifislax-ng, 2018-06-28:
As start idea , i think arround a configure option to enbale that part of code ..for example

./configure --enable-mac

Im very happy and zero problems , using the first option i mentioned time ago

pcap_set_rfmon(handle, 1); --> pcap_set_rfmon(handle, 0);

probably good idea ,default to 0 , and if enable-mac , turn to 1 this part.

Its only little idea, simply , but zero problems , and the best thing is never see monX in linux.

wifislax-ng, 2018-06-28:
After test kcdtv suggest and inflator, results is monX everywhere.

Think arround other tools , like airgeddon , minidwep , inflator , ... or a big number of scripts written from users to automated works.

wifislax-ng, 2019-03-31:
Hi again , i test the latest commit , the issue with multiplication of monitor interfaces is FIXED NOW.

THANKs!

rofl0r, 2019-03-31:
did you test the master branch or the apple_only_rfmon of the above PR ?

wifislax-ng, 2019-03-31:
> did you test the master branch or the apple_only_rfmon of the above PR ?

1.6.5_48a0a8b

wifislax-ng, 2019-03-31:
Thats funny , the apple branch works perfectly , no more monX everywhere.

The master branch , are bad.

rofl0r, 2019-03-31:
ok, i merged the PR into master now, so this can be closed.

closed 2019-03-31 by rofl0r