t6x/reaver-wps-fork-t6x #205: sanitize_string()  is incompatible with Chinese characters.

ChunshengZhao, 2018-02-27:
root@kali:~/test# ./chinese_ssid
Hex:
0x33,0x36,0x30,0xe8,0xa1,0x8c,0xe8,0xbd,0xa6,0xe8,0xae,0xb0,0xe5,0xbd,0x95,0xe4,0xbb,0xaa,0x2d,0x31,0x31,0x32,0x32
Normal:
360行车记录仪-1122
Abnormal:
360\\xe8\\xa1\\x8c\\xe8\\xbd\\xa6\\xe8\\xae\\xb0\\xe5\\xbd\\x95\\xe4\\xbb\\xaa-1122

[chinese_ssid.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/1762368/chinese_ssid.zip)

rofl0r, 2018-02-27:
well, we have a dilemma here. the sanitize_string() function was added so an attacker cannot inject terminal control codes by abusing the essid field to mess up your terminal or worse, write arbitrary files or execute code: https://www.proteansec.com/linux/blast-past-executing-code-terminal-emulators-via-escape-sequences/

a thinkable solution to get both security and chinese characters could be to check if the essid is valid UTF-8. but then how can i know if the terminal actually displays UTF-8 or ASCII.

this is a hard problem, and i went for an easy fix.

if you have better ideas let me know.

kcdtv, 2018-02-27:
>  how can i know if the terminal actually displays UTF-8 or ASCII.

Isn't the "echoed" value of `$LANG` enough? 

> if you have better ideas let me know.

In the worst case we could add a tag in wash to enable special characters and we warn the user in the option description that it might be risky.  That is not the best but does not require too much effort and seems correct as a security policy: User can take risk if he wants, he is warned and this is not the default behavior

rofl0r, 2018-02-27:
>>    how can i know if the terminal actually displays UTF-8 or ASCII.

> Excuseme if that is stupid... smiley_cat Isn't the "echoed" value of $LANG enough?

not really. in my case LANG is not even set at all. the behaviour can be affected from a number of variables: 1) environment variables 2) default locale 3) libc 4) shell 5) terminal

this is a very non-trivial topic, and i have so far successfully avoided to dive deep into it.

> In the worst case we could add a tag in wash

you mean yet another command line option ? `:-p`

kcdtv, 2018-02-27:
Yes, I mean another option... As we eliminated -o option there is at least one letter left... "o" like "Original bad essid" :smiley_cat: as it is well known that ssid should use no space or special character, just basic latin alphabet and numbers. But if you/me/somebody finds a nicer way, for sure I am all hear.

rofl0r, 2018-02-27:
yeah, let's first hear if @ChunshengZhao has some ideas, and if he insists having the chinese letters in his terminal.

btw i'm not a big fan of option letter recycling, at least not so quick after they've been removed. it's better if some time passes so ppl that used `-o` will notice it's no longer there... rather than doing a completely different thing without them noticing.

ChunshengZhao, 2018-02-28:
Do not care about the terminal character set, if the correct utf-8 string output, otherwise execute sanitize_string (). Maybe add “-utf8” option? Have a attacker’s code is in the utf8 format?

Sorry my english.

ChunshengZhao, 2018-02-28:
Like this:

root@kali:~/test# ./utf8_ssid 
Normal:
360行车记录仪-1122
Abnormal:
\\x8060\\xe8\\xa1\\x8c\\xe8\\xbd\\xa6\\xe8\\xae\\xb0\\xe5\\xbd\\x95\\xe4\\xbb\\xaa-1122

[utf8_ssid.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/1766970/utf8_ssid.zip)

ChunshengZhao, 2018-03-01:
[image]

[verifyssid.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/1769832/verifyssid.zip)

VerifySSID function (UTF-8 supported):
https://github.com/ChunshengZhao/verifyssid

rofl0r, 2018-03-02:
nice work. what's your plan with the function ?

ChunshengZhao, 2018-03-02:
If the development team think the verifyssid() is safe.
easy way :
add "if(verifyssid(s)) return s;" in your sanitize_string().

rofl0r, 2018-03-02:
i don't know if it is safe. how did you end up with it ? did you read the UTF8 specificiation, or study some validation code in detail ?

ChunshengZhao, 2018-03-02:
https://en.wikipedia.org/wiki/UTF-8
UTF8 string verification process is correct.
The only hidden danger: a malicious code matching UTF8 encoding rules.
The Reaver does not depend on the SSID, so this is an unnecessary patch.

ChunshengZhao, 2018-03-02:
@rofl0r 
New requests：
https://github.com/t6x/reaver-wps-fork-t6x/pull/206

commit e13598ae (2018-03-06): add option -U/--utf8 to opt out of ESSID sanitization

if this option is used and the ESSID passes a UTF-8 validity check,
the ESSID will be displayed unmodified. this is potentially dangerous
on terminals not supporting UTF-8 when an attacker embeds terminal
escape sequences.

closes #207
closes #206
closes #205
closed 2018-03-06 by rofl0r