t6x/reaver-wps-fork-t6x #209: Wash displays no output - OpenWRT CC (ar71xx - MIPS)

adde88, 2018-03-19:
I've compiled the latest version 1.6.4 for OpenWRT.
Basicly, wash displays only this:
```
BSSID               Ch  dBm  WPS  Lck  Vendor    ESSID
--------------------------------------------------------------------------------
[!] Found packet with bad FCS, skipping...
```
There's no "header", and no output. Using the **-F** argument does not help.

I have manually compiled different versions as explained in this post:
https://github.com/t6x/reaver-wps-fork-t6x/commit/664a4b3275f0e49124ec5139c032efbc9ed277ed

The issue started to happen on all versions i've tested on OpenWRT after  7 nov 2017.

I also want to point out that i have not used bisect to pinpoint where the bug occurred.
I've manually compiled different versions until i first noticed it happening.
Just wanted to inform you that some devices are having issues with the latest release.

I'll update this if i find any new information regarding the issue.

rofl0r, 2018-03-19:
fixed. since you're obviously capable of building from git for your mips device, it's still beyond me why you weren't able to build the 6 commits which git bisect would have shown you and tested those.

closed 2018-03-19 by rofl0r

adde88, 2018-03-21:
I have no previous experience with bisect, so i had to read up on it.
When i atttempted to use it it didn't present me with any specific commits to build.
I had to mark each commit "bad" or "good" after compilation.
In the end the SDK wasn't very "compatible" with bisect (it deleted the repo-information after each compilation. Basicly resetting the bisect each time.)

Progress halted a bit there. There's probably a way to go about it, but i couldn't think of one, sorry if this bisect issue annoyed you.
And thanks for fixing the issue 👍

commit 9bae55bd (2018-03-19): fix logical bug in endianness.h causing big endian breakage

closes #209

since OP didn't deliver `git bisect` results, i did it myself.