t6x/reaver-wps-fork-t6x #228: reaver error openwrt LEDE Reboot 17.01.4

tripplex95, 2018-03-30:
[+] Switching mon0 to channel 153
[+] Waiting for beacon from 60:19:71:B6:3F:35
Segmentation fault
 i dont know what is causing the problem please whats the step in resolving this issue i am using a nanobridge m5 with ar71xx radio

soxrok2212, 2018-03-30:
This does nothing to help us diagnose your problem. Are you using the latest build? Did you compile from source or grab from OpenWrt repo? What's your command? What is the rest of the output?

tripplex95, 2018-03-30:
The above is using reaver 1.6.4
i dont know what is causing the problem please whats the step in resolving this issue i am using a nanobridge m5 with ar71xx radio

downloaded openwrt LEDE Reboot 17.01.4 from openwrt repo tried stable and also snapshot same result

rofl0r, 2018-03-31:
is that device big endian ? if so you will have to use latest git

after you have compiled latest git, if you still experience a segfault, run wash/reaver with gdb: `gdb --args ./wash --options --here` then type `r` to run.
when it crashes type `bt` and paste output here.

tripplex95, 2018-03-31:
i am anew user not to knowledgeable when it come to openwrt but i am gonna try though and see what happens.

rofl0r, 2018-03-31:
in case it's possible to use a precompiled 1.6.3 you could also try that. the big endian bug was introduced after that.

tripplex95, 2018-04-01:
Send me a link to download please along with instructions

rofl0r, 2018-04-01:
sorry, this is no kindergarten here.

tripplex95, 2018-04-01:
LOL!!!

soxrok2212, 2018-06-04:
No activity in two months, closing for now.

closed 2018-06-04 by soxrok2212