t6x/reaver-wps-fork-t6x #233: WPS: Unexpected state (15) for receiving M1 / Realtek cannot be cracked

mlg556, 2018-04-19:
I have come across a Realtek access-point which cannot be attacked by pixiewps using Reaver v1.6.4. Using the triple verbose option, I have seen that there is a loop happening like this:
```
[+] Received M1 message
WPS: Processing received message (len=444 op_code=4)
WPS: Received WSC_MSG
WPS: Unsupported attribute type 0x1049 len=6
WPS: Parsed WSC_MSG
WPS: Received M1
WPS: Unexpected state (15) for receiving M1
WPS: returning
[+] Received M1 message
```
which results in a not found WPS pin. I am also posting this because the tool somehow found my data "interesting". Here is the full output:

```reaver -i wlan1mon -b D8:50:E6:5C:85:DC -vvv -t 40 -K -N 

Reaver v1.6.4 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Waiting for beacon from D8:50:E6:5C:85:DC
[+] Switching wlan1mon to channel 1
[+] Received beacon from D8:50:E6:5C:85:DC
[+] Vendor: RealtekS
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
     31 32 33 34 35 36 37 30                           12345670        
WPS: Selected registrar information changed
WPS: Internal Registrar selected (pbc=0)
WPS: sel_reg_union
WPS: set_ie
WPS: cb_set_sel_reg
WPS: Enter wps_cg_set_sel_reg
WPS: Leave wps_cg_set_sel_reg early
WPS: return from wps_selected_registrar_changed
[+] Trying pin "12345670"
send_packet called from deauthenticate() 80211.c:326
send_packet called from authenticate() 80211.c:357
[+] Sending authentication request
send_packet called from associate() 80211.c:410
[+] Sending association request
[+] Associated with D8:50:E6:5C:85:DC (ESSID: SOME_NAME)
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
[... 201 lines cut ...]
     52 65 61 6c 74 65 6b 20 53 65 6d 69 63 6f 6e 64   Realtek Semicond
     75 63 74 6f 72 20 43 6f 72 70 2e                  uctor Corp.     
WPS: Model Name - hexdump_ascii(len=11):
     41 44 53 4c 20 52 6f 75 74 65 72                  ADSL Router     
WPS: Model Number - hexdump_ascii(len=13):
     45 56 2d 32 30 30 36 2d 30 37 2d 32 37            EV-2006-07-27   
WPS: Serial Number - hexdump_ascii(len=15):
     31 32 33 34 35 36 37 38 39 30 31 32 33 34 37      123456789012347 
WPS: Primary Device Type: 6-0050F204-1
WPS: Device Name - hexdump_ascii(len=21):
     41 44 53 4c 20 52 6f 75 74 65 72 2f 4d 6f 64 65   ADSL Router/Mode
     6d 20 49 47 44  
```
or maybe
```
WPS: Enrollee Authentication Type flags 0x21
WPS: No match in supported authentication types (own 0x0 Enrollee 0x21)
WPS: Workaround - assume Enrollee does not advertise supported authentication types correctly
WPS: Enrollee Encryption Type flags 0x9
WPS: No match in supported encryption types (own 0x0 Enrollee 0x9)
WPS: Workaround - assume Enrollee does not advertise supported encryption types correctly
WPS: Enrollee Connection Type flags 0x1
WPS: Enrollee Config Methods 0x686 [Ethernet] [Label] [PBC]
WPS: Prefer PSK format key due to Enrollee not supporting display
```

kcdtv, 2018-04-21:
FYI the message is issued by pixiewps itself and you can use a little script to extract the datas, see [Data gathering for pixiewps (pixie dust attack)](https://forums.kali.org/showthread.php?38127-Data-gathering-for-pixiewps-(pixie-dust-attack)) But many thanks anyway for your feedback. :smiley_cat:

rofl0r, 2018-04-22:
```
executing pixiewps -e d0141b15656e96b85fcead2e....
```

when you get that message you can copy/paste the entire pixiewps command into the shell and add --force

soxrok2212, 2018-04-22:
Realtek has actually started implementing patches. See http://www.belkin.com/us/support-article?articleNum=24890 and look at patch notes.

rofl0r, 2018-05-09:
@kcdtv if you have time the next couple days i guess we should push out a new release since this version is "stale" since a while

kcdtv, 2018-05-09:
@rofl0r I will be around the next couple of days. Let's do it!