t6x/reaver-wps-fork-t6x #234: reaver Not available！！

wihack, 2018-05-06:
When I used ubuntu 16.04: sudo reaver -i wlan0mon -b BC:D1:77:17:7C:B4 -p 00189743 , it did not show WPA-psk password.
] Waiting for beacon from XX:XX:XX:17:7C:B4
[+] Received beacon from XX:XX:XX:17:7C:B4
[+] Vendor: AtherosC
[+] Associated with XX:XX:XX:XX:7C:B4 (ESSID: Honey)
[+] Associated with XX:XX:XX:XX:7C:B4 (ESSID: Honey)
[+] Associated with XX:XX:XX:XX:7C:B4 (ESSID: Honey)
[+] Associated with XX:XX:XX:XX:7C:B4 (ESSID: Honey)
[+] Associated with XX:XX:XX:XX:7C:B4 (ESSID: Honey)

But I don't have these problems on kali linux, I want to solve this problem with ubuntu.

rofl0r, 2018-05-06:
ok, send me your router and your computer and i'll tell you whats wrong.

wihack, 2018-05-07:
OS:Ubunutu 16.04  router:TP-LINK

soxrok2212, 2018-05-07:
What is the model of the router? What is your wireless card? Can you log into the router and verify that WPS is both enabled and configured? Move closer (within -40 RSSI) and see if interference is a problem.

wihack, 2018-05-07:
I run perfectly on kali linux. Only ubunutu can not. The network cardl has tried several, but it is not possible, only the kali can be no error on it, but I want to run on ubuutu。
WPS settings are all turned on, router operation is no problem

soxrok2212, 2018-05-07:
Please answer all the questions I asked :)

wihack, 2018-05-07:
OK, my router model is: TP-LINK847N, network card is AWUS036NH and IMC Networks and netgear WG11V3. The router is my own and there is no signal interference. What driver do I need?

kcdtv, 2018-05-07:
**Ubutnu 16.04** uses kernel **4.4** which is very problematic. It has the well known "ralink digression" bug that affects all ralink USB chipset. This kernel version is also problematic with rtl8187l. So that the version of linux kernel that you better avoid.   
You have 3 solutions:    
 *  1) Update your distribution: Ubutnu 18.04 is out since some weeks. It is a LTS version too. I would say it is one of the best releases since quite a time. If your computer is 64bits and has 2GB or more, you should try it. 
 *  2) Install a newer kernel. For example:
`sudo apt install linux-headers-4.15.0-13-generic linux-image-4.15.0-13-generic linux-image-extra-4.15.0-13-generic linux-image-4.15.0-13`  
  *  3) Patch your actual kernel. Notice that Ubuntu gives you the possibility to live patch your kernel (you do not need reboot to see the effect) which is a cool feature: [canonical live patch](https://www.ubuntu.com/server/livepatch). You have to register to get it, you can use the feature in 3 different computers with the free version.

wihack, 2018-05-09:
Wow, thank you, you are amazing, I solved the problem by updating the patch!

closed 2018-05-09 by soxrok2212