t6x/reaver-wps-fork-t6x #238: Wash - Vendor Ubiquiti Networks Inc (44D9E7)

DJDan, 2018-05-26:
3 seperate AP networks that begins with "F0:9F:CC:xx:xx:xx" appears in wash as "Unknown" and seems to have a VENDOR ID of 44D9E7 (Ubiquiti Networks Inc)

Should this be added under the WIFI CHIPSET as "Ubiquiti Networks Inc" or is it using something else?
(eg; Atheros/RalinkTE/Broadcom)

NOTE:
Using wifi explorer for osx...  it seems to show 
vendor specific: ubiquiti then 
element id: 221
length: 5 bytes
oui: 44-D9-E7
vendor specific type: 65
vendor specific data: 86
(for 3 seperate AP which has the same SSID...)

If Wash can be updated to include VENDOR ID: 44D9E7 let me know and I will retest with the changes.

Thanks

rofl0r, 2018-05-26:
ubiquity is not a chip vendor.

closed 2018-05-26 by rofl0r

kcdtv, 2018-05-26:
I guess it is an atheros chipset. But we can't simply decrete that every ubiquiti has an atehros, Even if it is true now, they could perfectly use a quantena or mediatek chipset in the next model. A few vendors/products do not allow to remotly know the wifi chipset, like the sony Bravia TV. In this case they hace to be shown s "unknown", If we are not 100% sure it means that we do not know.