t6x/reaver-wps-fork-t6x #239: wash not correctly showing WPS enabled APs with missing "wps_version"

He1my, 2018-05-31:
My AP has WPS enabled, however Wash doesn't correctly show it

wash -c 11 -a -s  -j -i wlan4 

{"bssid" : "10:FE:ED:CB:0E:63", "essid" : "Vodafone ",     "channel" : 11, "rssi" : -86, "vendor_oui" : "000C43", "wps_state" : 2, "wps_locked" : 2, "wps_manufacturer" : "Ralink Technology, Corp.", "wps_model_name" : "Ralink Wireless Access Point", "wps_model_number" : "RT2860", "wps_device_name" : "RalinkAPS", "wps_serial" : "12345678", "wps_uuid" : "bc329e001dd811b2860110feedcb0e63", "wps_response_type" : "03", "wps_primary_device_type" : "00060050f2040001", "wps_config_methods" : "0004", "wps_rf_bands" : "00", "dummy": 0}                                                                                                                                   

wash -c 11 -a -s  -i wlan4 
BSSID                        Ch  dBm  WPS  Lck  Vendor    ESSID
--------------------------------------------------------------------------------
10:FE:ED:CB:0E:63   11  -86                       RalinkTe  Vodafone 

comparing the json output with other correctly identified APs, the "wps_version" is missing from the json output 
it seems Wash for some reason fails to detect it has WPS because it wasn't able to get the "wps_version" for some reason

kcdtv, 2018-05-31:
version is not detected with this parameters. Could you capture the whole probe and send it to us in order to see what is not working? It looks like the probe parameter were not correctly entered by the manufacturer which "breaks" the parsing. Thanks for your report.

He1my, 2018-05-31:
can you direct me to instructions on how I would go about capturing the the whole probe correctly if the attached pcap file is not correct, 

I tried to capture it using wireshark,
using wash with the --file parameter and the attached file shows the issue.
in wireshark it shows the wps version: 0x00 , maybe that is the problem?!

[10:fe:ed:cb:0e:63.pcap.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/2060015/10.fe.ed.cb.0e.63.pcap.zip)

kcdtv, 2018-05-31:
i removed the cap file in order to preserve your privacy... 
Better if you send it to me iby mail instead. I think you can do it from github. Otehrwise you can send it to kcdtv@wifi-libre.com 
I can't see any probe answer from the vodafone device.
I think it is better if you use tcpdump or airodump-ng
You colud  do something like this in one shell
`wash -i <interface> -c <channel> -s -n 100 `
and in another shell something like this 
`tcpdump -i <inetrface> -w <name for a file>`
cheers and thanks again :smile_cat:

He1my, 2018-05-31:
I had changed the ESSID today sorry about that , the BSSID is the same

kcdtv, 2018-05-31:
Okay... so we do not need any more cap, i see a full probe asnwer from the device. do not grab any more traffic, we have got it. I was on my way to bed actualy, i'll check that with r0flor and soxrock2212 tomorow, Maybe r0flor will pass by later... anyway, we wil tell you something for sure tomorow.  thanks again :)

He1my, 2018-06-01:
Did some more invastigation into this and found 5 other APs with the exact same issue, 
airodump with --wps option shows them correctly with the version being 0.0 after a while of scanning 
I think this issue may have an effect on reaver as well, when tried to use reaver with any of those APs with version 0.0 it just hangs after getting the beacon and doesn't associate

kcdtv, 2018-06-03:
> --wps option shows them correctly with the version being 0.0 after a while of scanning

There is no such thing as WPS version 0.0 so that not what i would call "correct". First version of WPS is WPS 1.0.  
it looks like a miss-configuration (voluntary or not) regarding the probe. Can you tell us which model of TP-link router is it?

rofl0r, 2018-06-04:
even though that router model is ignoring the correct way of doing things, there's a "lock" in the reaver code that prevents it from being used with non-WPS-enabled routers. this annoyed me when i tried the rewritten association code... i think we should get rid of this "lock" and only print a hint that the router is not WPS enabled when we're past the association stage.

the "lock" was probably put in place to prevent tries against a router that's been locked out.

generally i think it displaying version `0.0` for such routers is an OK choice, but it depends on implementation details if it's easy to support it...i'm quite certain the value 0 is already used to mean "uninitialized".

anyway, let's wait for more details about the affected router model(s).

He1my, 2018-06-08:
The router is a TP-Link TD-W8951ND

kcdtv, 2018-06-08:
There are 6 versions of this device, the last one (version 6), uses a mediatek wifi chip so we can discard it. 

> the "lock" was probably put in place to prevent tries against a router that's been locked out.

If that is  the case (but is it? there is the tag `"wps_locked"` dedicated to that and i am pretty sure  that it is the one used by reaver to detect if a router is locked or not) there is no real need to change anything as you can use the **-L** option to disable it 
`-L, --ignore-locks              Ignore locked state reported by the target AP`

@He1my: Could you check if your firmware wasn't updated by vodafone recently ( i assume this model is given by your ISP as you found so many around)

He1my, 2018-06-09:
the firmware is vodafone branded as evident by the vodafone branded web interface, no idea if it was recently remotely updated by vodafone or not, I no longer have the router with me, note that the vodefone router with the wps version misconfiguration showed correctly on android and windows devices as having wps enabled, and no issues regarding the wps function when used with them, could that be some kind of an attempt by vodafone to "secure" wps on their older routers?

kcdtv, 2018-06-10:
> the firmware is vodafone branded as evident by the vodafone branded web interface

Thanks for your feedback. I  guess they do the same everywhere and the web interface is quite limited with a real "root" password that they do not give to final customer and keep for themselves. That the way they do things in Spain since some years now with a unified interface for all devices. I was saying in my previous comment that we could forget about version 6 as it has a mediatek chipset. I forgot that your clearly says to have a **RT2860** wifi chipset so we can be 100% sure about that. That surprised me: i thought this chipset was just used in some PCI and USB client cards. Looking in the datas i collected during the past years i found others devices with this chipset (ISP is orange in this case) where the probes are correctly redacted and the WPS attack could be done successfully using pixiewps or the "ComputePIN" algorithm (end bSSID converted form hexadecimal to decimal)

> and no issues regarding the wps function when used with them

If you can connect with android or windows using the WPS PIN than it can be fixed. Would you be so kind to follow our debug guide and  pass us the cap grabed and the full verbose reaver stdout? [Troubleshoting](https://github.com/t6x/reaver-wps-fork-t6x/wiki/Troubleshooting)

He1my, 2018-06-10:
Sadly I no longer have the router to do further testing, gave it to a friend to dump its flash memory, do some firmware analysis and see if we can find out the root password and how the pin is calculated 

pixiewps didn't work with it when I tried it 

you mentioned 

> the "ComputePIN" algorithm 

does reaver have something similar to the wps pin generator in https://3wifi.stascorp.com/wpspin
https://github.com/binarymaster/3WiFi/blob/master/wpspin.html

rofl0r, 2018-06-11:
> "ComputePIN" algorithm

aka *zhao*, right ?

> does reaver have something similar to the wps pin generator

wash's -j json output was designed so 3rd-party scripts can easily grab the fields from WPS data in probe response to output a list of probable candidates. there does not exist a public script making use of it, yet, but it's trivial do do in python, ruby, etc.

kcdtv, 2018-06-12:
> does reaver have something similar to the wps pin generator

  Reaver included ComputePIN algorithm (Wrongly named and without the adequate references, that was a violation of GPL3 ) and included the algorithms discovered by Craig Heffner [pingens by devttys0](https://github.com/devttys0/wps/tree/master/pingens)
Bessie the "ethical" issue,  the way it was done was not satisfying at all: The default PIN was generated after sending a full PIN despite the fact that the strings required are available in beacon-probe and you may lock the router for nothing. You could not use the generator by entering manually bssid or essid neither, the code were not working correctly after some updates and for that reasons we dropped them.  The three algorithms are very well known and used in all "third party scripts" that you may find around 

> aka zhao, right ?

It was published by Zhao Chunsheng in early 2012, yes.

> there does not exist a public script making use of it, yet, but it's trivial do do in python, ruby, etc.

Wrong! [nmk.sh](https://github.com/kcdtv/nmk/blob/master/nmk.sh) :smile_cat:

rofl0r, 2018-06-14:
i guess we could assign it -1 and later before printing it out change it back to 0 when it's -1

He1my, 2018-06-14:
> i guess we could assign it -1 and later before printing it out change it back to 0 when it's -1

but the code checks if the WPS version is greater than 0, how would assigning it -1 would work?

rofl0r, 2018-06-14:
well, then change it to != 0

kcdtv, 2018-06-18:
It doesn't sounds correct to me to say "WPS version 0" as there is no version 0 of WPS.

commit 78651c21 (2018-06-19): wash: display APs with broken WPS version 0.0 field

closes #239

note: locked field != 2 is tested because any AP gets 2 assigned as
default value even if WPS is not active.
closed 2018-06-19 by rofl0r