t6x/reaver-wps-fork-t6x #240: WPS: Full PIN information revealed and negotiation failed

rafaleon69, 2018-06-02:
Hello pin crackers!

For the last few days I tried to crack I tried to crack this PIN running on a router RealtekS 2.0. I am using latest 1.65 reaver. It seems like the full PIN has been found, please see description below:

WPS: Unsupported attribute type 0x1049 len=6
WPS: Parsed WSC_MSG
WPS: Received M5
WPS: Unexpected state (18) for receiving M5
WPS: Incorrect Authenticator
WPS: Processing decrypted Encrypted Settings attribute
WPS: E-SNonce1 - hexdump(len=16): 2e 19 9d 89 d8 17 22 6c c5 74 16 40 55 c5 ad 04
WPS: Enrollee proved knowledge of the first half of the device password
WPS: WPS_CONTINUE, Freeing Last Message
WPS: WPS_CONTINUE, Saving Last Message
WPS: returning
[+] Received M5 message
WPS: Processing received message (len=120 op_code=4)
WPS: Received WSC_MSG
WPS: Unsupported attribute type 0x1049 len=6
WPS: Parsed WSC_MSG
WPS: Received M5
WPS: Unexpected state (18) for receiving M5
WPS: Incorrect Authenticator
WPS: Processing decrypted Encrypted Settings attribute
WPS: E-SNonce1 - hexdump(len=16): 2e 19 9d 89 d8 17 22 6c c5 74 16 40 55 c5 ad 04
WPS: Enrollee proved knowledge of the first half of the device password
WPS: WPS_CONTINUE, Freeing Last Message
WPS: WPS_CONTINUE, Saving Last Message
WPS: returning
[+] Received M5 message
WPS: Processing received message (len=66 op_code=3)
WPS: Received WSC_NACK
WPS: Unsupported attribute type 0x1049 len=6
WPS: Enrollee terminated negotiation with Configuration Error 18
[+] Received WSC NACK
WPS: Building Message WSC_NACK
WPS:  * Version
WPS:  * Message Type (14)
WPS:  * Enrollee Nonce
WPS:  * Registrar Nonce
WPS:  * Configuration Error (0)
[+] Sending WSC NACK
send_packet called from send_msg() send.c:116
WPS: Full PIN information revealed and negotiation failed
WPS: Invalidated PIN for UUID - hexdump(len=16): 63 04 12 53 10 19 20 06 12 28 48 3c 0c 92 43 50

I am running a command:

./reaver -b xx:xx:xx:xx:xx:xx  -p 97423171 -c 6 -i wlan1mon -vvv -N -T 2.00 -r 1:5 -w

without N option the transaction would just fail, I can only reproduce this result above by adding -N. I tried using this method to connect with the router to get the PSK: 
https://miloserdov.org/?p=138
But It would not allow me to connect either.

I am out of ideas at the moment, any suggestions ?

rofl0r, 2018-06-02:
can you try earlier versions, 1.6.4, 1.6.3 etc and report back ? you should also capture what happens with wireshark/tcpdump/airmon-ng and send it to @kcdtv : kcdtv@wifi-libre.com so we can analyze why tehre's no progress after M5.

rafaleon69, 2018-06-03:
Done, email sent. Please do let me know if you need anything else.

kcdtv, 2018-06-03:
Your packet does not contain the recieved message. Could you send me the original *cap, without any filtering, i promise we erase them when we solve the issue ;) 
cheers

rafaleon69, 2018-06-03:
capture updated, check your email. thanks for looking into that.

kcdtv, 2018-06-03:
Got it! Thanks to you :smiley_cat:

rafaleon69, 2018-06-03:
Any suggestions of what can I try to do with that ?

rafaleon69, 2018-06-04:
No go gentlemen, I got close enough to get 47db and still the same situation. What the duce ?

rafaleon69, 2018-06-04:
aireplay-ng -a xx:xx:xx:xx:xx:xx -9 wlan0
06:43:38  Waiting for beacon frame (BSSID: xx:xx:xx:xx:xx:xx) on channel 6
06:43:47  Trying broadcast probe requests...
06:43:47  Injection is working!
06:43:49  Found 1 AP 

06:43:49  Trying directed probe requests...
06:43:49  xx:xx:xx:xx:xx:xx - channel: 0 - 'TALKTALK924348'
06:43:54  Ping (min/avg/max): 2.652ms/4.201ms/8.298ms Power: -57.40
06:43:54   5/30:  16%

hmm a bit low...

FYI I also specified the pin with bully bit it said the pin was bad and it went to try out next one.

I will work on the way to increase the quality of injection...

rafaleon69, 2018-06-04:
But still wouldn't injection test indicate how well I can send the packets to the router which is what TX - transmit - stands for? RX - recieve would indicate a communication back to me, which is something I try to test.. Please do correct me if I am wrong.

kcdtv, 2018-06-04:
Injection test is probe answer-probe response (i am pretty sure but not 100%) If you get 30/30 - 100% it means that you are in correct TX & RX conditions 
16% is definitely too low, 90% should be enough, 

> FYI I also specified the pin with bully but it said the pin was bad and it went to try out next one.

That not surprising as you never get to M7... if you get the M5 it means the first half is correct. You need to get the M7 to know if the second half is correct. reaver was desined to test a single pin when you use the -p option (which i personally do not like at all and always found a very bad approach, but that was the conception from craig heffner). A good habit is to use -g 1 when you use -p, in order to no enter in a stupid loop when reaver keep on trying the same PIN even if it is not correct. Notice that the use of -p option does not allow to keep a record from the session neither... That also something i do not like at all. what i do in this case is to redact my own wpc file and use the -s option to launch it. like this you can keep record of the session.

nuncan, 2018-11-23:
Never liked AC1200 (trust me I wanted to). Get yourself a AWUS036NHR and a Yagi. And if law allows, adjust regdb