t6x/reaver-wps-fork-t6x #241: ¿Github or not Github?

kcdtv, 2018-06-06:
Hi people!
  You may have notice this "nice" banner in the top of Github with the announcement of "Mircosoft is acquiring Github!"
  The public enemy of free software, the most hegemonic monopolistic ever seen firm in this "industry"  is now claiming to be a "free software lover". 
  It is quite insulting and grotesque and you would have to be dumb to buy that crap: If microsoft try to change its image it  is simply because they lost their fight. We are not in the 90'-2000' anymore when they could do what they want.  Their OS is barely used in mobile devices, android got it all with mac. They never managed to impose themselves on the server market neither, linux and unix rules here and in this "all clouding time" it is a big failure. They simply do not exit on the IoT market which is the furture (i would say it is even the present). The time when Bill Gate was buying everything in the sillicon valey just to close it in order to kill any concurrence is well over. They desperately need a new strategy if they want to be a leader and to be something else than the OS of the old PC for old-timers.  
  I am personally not preoccupied by this. I don't think that they will fucked it up with github, they spent 7,5 billions dollars in order to change their image by buying github and it would be very bad for them if the community turns it back on github.   Saying so some major free softwares such as gnome and gimp decided  to migrate to gitlab for that reason.  I would not take such a drastic measure. But i would argue that it could be a good thing to maintain reaver also in gitlab. If "something goes wrong with github" we would be ready to migrate and the fact that gitlab gets a lot of repositries (10 time more than usual since this baners is here) sends a clear message to Satya Nadella: Do not fuck it up! 
 What do you think?

binarymaster, 2018-06-06:
> Mircosoft is acquiring Github!

I don't see a problem there. And even more... I'm not planning to move my projects elsewhere, unless MS would do something really bad.

Remember: Git is distributed VCS.

kimocoder, 2018-06-06:
I'm not moving mine either. My repos stays

soxrok2212, 2018-06-06:
Not to mention, GitLab now offers premium plans for open source software.

Personally, I think I will move to GitLab.

nico202, 2018-06-06:
GitHub has never been free software, so it was a mistake to be here in the first place. I'm hosting gitea by myself and will leave gh "soon"

rofl0r, 2018-06-08:
i think we should get @t6x to make an account over at gitlab or notabug, which we then can announce here as the backup account should microsoft decide that they don't want to host offensive penetration testing tools. that said, making a repo over there already and syncing it consistently by hand is tedious. if there's a way to have it mirrored automatically, i'm all for it, but if not we can upload the repo *iff* things go wrong here.

closed 2018-07-05 by kcdtv