t6x/reaver-wps-fork-t6x #242: Enhancement request , -V  , to print version only.

wifislax-ng, 2018-06-07:
Hi , i think this is a little but recommended enhancement , only add -V to see "version" in reaver and wash , i put -V cuase -v ARE USED IN REAVER FOR VERBOSE  , that little request make more easy extract version . for other people writing some bash , or python script.

Its practivally universal in linux a command to see verions on programs , normally -v . or -V 

Thanks in advance.

rofl0r, 2018-06-08:
you can always use --help to get the help screen which has the version number. adding a new flag won't help you since the old versions don't support it anyhow, and my feeling is that there will not be a lot of development happening here in the future, as WPA3 and a new (supposedly secure) WPS standard is expected to hit the markets this year.

closed 2018-06-08 by rofl0r

rofl0r, 2018-06-08:
much likely. yet many people will still continue to use their existing routers, so maybe reaver will survive another decade...