t6x/reaver-wps-fork-t6x #243: wash: Adding a descriptive array of the enabled WPS configuration methods to json output

He1my, 2018-06-11:
What are your thoughts about adding an easier method to output the WPS configuration methods in json,
for example 
`"wps_config_methods": "2084", "wps_config_methods_str": ["Label", "Push Button", "Virtual Display"],`

according to Wireshark the bit values for the WPS configuration methods are 
```
.... .... .... ...1	USB
.... .... .... ..1.	Ethernet
.... .... .... .1..	Label
.... .... .... 1...	Disply
.... .... ...1 ....	External NFC
.... .... ..1. ....	Internal NFC
.... .... .1.. ....	NFC Interface
.... .... 1... ....	Push Button
.... ...1 .... ....	Keybad
.... ..1. .... ....	Virtual Push Button
.... .1.. .... ....	Physical Push Button
..1. .... .... ....	Virtual Display
.1.. .... .... ....	Physical Display
```

rofl0r, 2018-06-12:
i've stated on several occasions that the existing json output will not be modified (apart from eventually adding new field), so users of it have the guarantee that their scripts (consuming the json) will keep working.

it's trivial to add code in your script that parses the raw config methods.

we could provide a tiny filter tool (in tools/) which takes the default wash output and modifies into the form suggested by you: `./wash -j -s -n 100 | ../tools/pimp_json.py | yourfinalscript.py`

He1my, 2018-06-12:
what is suggested here is adding a new field "wps_config_methods_str", I am, well aware that that there should not be any modification to existing fields

rofl0r, 2018-06-12:
PR welcome

DJDan, 2018-07-06:
This already exists in wash 1.6.5-git-1-g140507a
wash -j json output has wps_config_methods .

He1my, 2018-07-06:
> This already exists in wash 1.6.5-git-1-g140507a
> wash -j json output has wps_config_methods .

yes "wps_config_methods" is there, but that is not what this is about, it is about adding another field "wps_config_methods_str" that shows the enabled wps config method names instead of just the hex value to make is easier to understand