t6x/reaver-wps-fork-t6x #244: wash and reaver stuck in "waiting for beacon"

Jiyuu, 2018-06-12:
I have a laptop with a bcm4360 wireless chipset.
I installed the broadcom-wl drivers and enabled it's monitor&injection mode based on the instructions at https://wiki.archlinux.org/index.php/Broadcom_wireless#Setting_broadcom-wl_in_monitor_mode
following that i performed a check using the `airodump -i prism0` command, which seemed to suggest it was constantly capturing packets, identifying beacons, handshakes, and when supplied with the "--wps"  flag, it identified wps data as well.
using airodump to actually dump the packets data into a file and having wash use that file produced proper information about the networks, similar to what airodump was showing.

the issues are:
* wash does not work with the interface directly, it just does nothing.
* when supplying the info to reaver directly based on the file captures, it gets stalled at "waiting for beacon" and does nothing.

soxrok2212, 2018-06-12:
AFAIK, BCM4360 doesn’t not have injection.

rofl0r, 2018-06-12:
@Jiyuu maybe try if it works with older versions. there have been some changes in the code dealing with the setup of the monitoring device.

kcdtv, 2018-06-12:
the lack of injection would not explain the failure of wash. Also notice that you might need to run airodump-ng before using wash to "wake up" the interface. tehre is some agressive routine in airdoump-ng taht does not have wash and it happens to me to have to use that trick to have wash working.