t6x/reaver-wps-fork-t6x #245: WARNING: Receive timeout occurred while using REAVER and AIRCRACK-NG

danieleperera, 2018-06-17:
I enabled wps on my own router and I'm trying to crack it using reaver and aircrack-ng but I always get this warning.

I'm using awus036ach chipset Realtek RTL8812AU. Installed following instructions on https://github.com/aircrack-ng/rtl8812au

And I think it's working fine because when I try aircrack-ng test injection it works fine.

wlan0     IEEE 802.11  Mode:Monitor  Frequency:2.427 GHz  Tx-Power=18 dBm   
              Retry short limit:7   RTS thr:off   Fragment thr:off
              Power Management:off

**reaver -i wlan0 -b XX:XX:XX:XX:XX:XX --mac-changer --no-nacks --win7 --no-associate -vv**
 
**aireplay-ng wlan0 -1 120 -a XX:XX:XX:XX:XX:XX --ignore-negative-one -h XX:XX:XX:XX:XX:XX**

Reaver v1.4 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Waiting for beacon from XX:XX:XX:XX:XX:XX
[+] Switching wlan0 to channel 1
[+] Associated with XX:XX:XX:XX:XX:XX (ESSID: AAAAAAAAAAAA)
[+] Using MAC XX:XX:XX:XX:XX:XX 
[+] Trying pin 12345670
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred

Have you got any suggestions to resolve this problem?

soxrok2212, 2018-06-18:
First, you should try using reaver from this repository and not 1.4.

danieleperera, 2018-06-18:
@soxrok2212 okay I'll try and let you know. Thanks for the quick reply.

soxrok2212, 2018-06-18:
Reboot your router or disable the lock in the web GUI

rofl0r, 2018-06-18:
wtf is --mac-changer ? :D well, anyway, i guess this is solved and can be closed...

closed 2018-06-18 by rofl0r

danieleperera, 2018-06-18:
@rofl0r in the Reaver v1.4 I guess there was the option to change the mac address of the wlan interface which is trying to crack wps. It would be nice to have that option with this version too. Thanks indeed.

danieleperera, 2018-06-18:
https://github.com/gabrielrcouto/reaver-wps @rofl0r this is where I found the script.

rofl0r, 2018-06-18:
no, reaver 1.6.x (what you find here) is based directly on upstream reaver 1.4. there was no --mac-changer option in the original. the author of the mod you link to seems to have made this option because he didn't understand the cause of his connection problems.