t6x/reaver-wps-fork-t6x #246: lock-out

liklev, 2018-06-19:
hi, testing my huawei AP (tough little bastard) it goes on lock-down after 3 pin. 180 seconds still not enough delay between PINs. the issue is when i change mon0 MAC reaver does not get over sending M1 (received M1 - sending M2 -again receiving M1 instead of M3). As soon as i change back to factory MAC of my wifi all exchange ok (m1-m2-m3-m4) but the lock-down after 3rd pin. (either pixie dust does not harm this AP).
I have openwrt snapshot of 2018jun18. still confused with hardware . beside the mon0 have a br-lan, eth0, eth0.1, eth0.2 what all have the same MAC address. injection works perfectly (WLAN Hardware:
Qualcomm Atheros QCA9563, Qualcomm Atheros QCA9880).
so is it a reaver issue or i have to clear everything up to accept spoofed MAC?
thanks

kcdtv, 2018-06-29:
What you describe is the normal routine when you face a router with blocking countermeasure  (as expected with WPS 2.0)
 Changing the mac address would not have any effects: The router will anyway lock the protocol (for all _enrollees_) after three wrong first halfs PIN (regardless who sent the PIN). To say it in another form: The counting of wrong attempts is "global", the locking is "global" too.    
That leaves you with a few attempts that you have to use wisely.

kimocoder, 2018-06-29:
mdk4 has 1 or 2 attacks which may get the router out of lock state by flooding it

rofl0r, 2018-06-29:
> The counting of wrong attempts is "global", the locking is "global" too.

not only that, many routers save the locked-out state to their nvram, so after a reboot they're *still* locked, the only way to get them unlocked again is by doing a factory reset or by manually unlocking them in the web interface, if possible.

liklev, 2018-06-30:
funny thing i attempted a pixie dust with the proper exact pin of the router and it is so invulnerable that pixie reports "no pin found". ok, for the same proper pin reaver cracks out the wpa2 passphrase.
my huawei with wps2.0 probably even delay between pins wont work (10 mins ive tried max 6000secs).
and mdk3 flood with 200000+ clients was only slowing it down a bit.

closed 2018-07-05 by kcdtv