t6x/reaver-wps-fork-t6x #251: MacOS "Receive timeout occurred"

Tommrodrigues, 2018-06-30:
Hey there,

I'm pretty new to reaver and have it installed on my MacBook for penetration testing on my router. Before I start using reaver I make sure that I have enabled monitor mode on my wifi card using the command `sudo airport en0 sniff 1` Then I use the reaver command `sudo reaver -i en0 -b 34:57:60:B6:97:62 -vv -N -S -c 1 -t 10` in a separate window. However, the problem is that I keep revieving the error message "WARNING: Receive timeout occurred". I have experimented with a number of different flags in the reaver command but can't find anything that seems to get it to work. I have done some googling and found that others have had the same problem and I have tried all of the acclaimed solutions however this doesn't seem to help. I have a feeling this is because I am running it on a Mac rather than on Kali. Below is an example of one of my reaver sessions where you can see the error:

```
Tommys-MacBook-Pro:~ tommyrodrigues$ sudo reaver -i en0 -b 34:57:60:B6:97:62 -vv -N -S -c 1 -t 10
Password:

Reaver v1.4 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching en0 to channel 1
[+] Waiting for beacon from 34:57:60:B6:97:62
[+] Associated with 34:57:60:B6:97:62 (ESSID: Rodrigues)
[+] Trying pin 12345670
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
^C
[+] Nothing done, nothing to save.
```

Please could someone give me some advice on how to fix this?

Thank you in advance for any help,
Kind regards,
Tom

P.S. I installed reaver with `brew install reaver` and it seems to be on the correct version as you can see from my example session. If you require any more information please do not hesitate to ask!

rofl0r, 2018-06-30:
> Reaver v1.4 WiFi Protected Setup Attack Tool

try again with latest git. current is 1.6.5. you gotta compile it from source.

Tommrodrigues, 2018-06-30:
Thanks for your reply, didn't realise we were on 1.6.5! Anyway, I've uninstalled the one I had from brew and have installed it from the repository instead. However, I still have no success. Furthermore, I am receiving other errors such as "Found packet with bad FCS, skipping...". Any ideas on what to do? I'll post my most recent session below:

```
Tommys-MacBook-Pro:src tommyrodrigues$ sudo reaver -i en0 -b 34:57:60:B6:97:62 -vv -d 10 -N -S -c 1 -t 10

Reaver v1.6.5-git-6-gf813196 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching en0 to channel 1
[+] Waiting for beacon from 34:57:60:B6:97:62
[+] Received beacon from 34:57:60:B6:97:62
[+] Vendor: RealtekS
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with 34:57:60:B6:97:62 (ESSID: Rodrigues)
[+] Sending EAPOL START request
[+] Received deauth request
[!] Found packet with bad FCS, skipping...
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received deauth request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received deauth request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received deauth request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received deauth request
^C
[+] Nothing done, nothing to save.
```

Thanks

rofl0r, 2018-06-30:
```
[+] Sending EAPOL START request
[+] Received deauth request
```
the router is sending deauths after your EAPOL START. either it's too far away or WPS functionality has been crippled by a firmware update (many ISPs have decided to do that rather than fixing their vulnerabilities).

Tommrodrigues, 2018-06-30:
Most likely the latter as I purposefully placed my laptop right next to my router to get the best chance. Shame but oh well. Strange thing is that when running `wash` it comes up as not being locked but I guess that if all they've done is crippled WPS functionality rather than fixing it properly then that explains it. So there's no way of getting around it then?
Thanks once again

Tommrodrigues, 2018-07-01:
The model is "VB Mitrastar" and the ISP is "Telefónica".

rofl0r, 2018-07-02:
if `wash -j -a -s -n 100 -i en0` looks like this
```
{"bssid" : "98:97:D1:XX:XX:XX", "essid" : "MOVISTAR_XXXX", "channel" : 1, "rssi" : -81, "vendor_oui" : "000C43", "wps_version" : 32, "wps_state" : 2, "wps_locked" : 2, "wps_manufacturer" : "MitraStar", "wps_model_name" : "MitraStar Wireless Access Point", "wps_model_number" : "RT7592", "wps_device_name" : "MitraStar DSL-100HN-T1v4", "wps_serial" : "12345678", "wps_response_type" : "03", "wps_primary_device_type" : "00060050f2040001", "wps_config_methods" : "210c", "wps_rf_bands" : "01", "dummy": 0}
```
then yeah, this model was vulnerable until roughly oct' 2017 but then got a firmware update that crippled WPS.

Tommrodrigues, 2018-07-02:
Just ran that command and got 

```
{"bssid" : "34:57:60:B6:97:62", "essid" : "Rodrigues", "channel" : 6, "rssi" : -32, "vendor_oui" : "00E04C", "wps_version" : 32, "wps_state" : 2, "wps_locked" : 2, "wps_manufacturer" : "MistraStar Corp.", "wps_model_name" : "RTL8xxx", "wps_model_number" : "EV-2010-09-20", "wps_device_name" : "HGW-500BNA-QCv2", "wps_serial" : "123456789012347", "wps_uuid" : "63041253101920061228345760b69762", "wps_response_type" : "03", "wps_primary_device_type" : "00060050f2040001", "wps_config_methods" : "0000", "dummy": 0}
```

Which is pretty similar to what you posted so I'm assuming that is the case then.

rofl0r, 2018-07-02:
i've never seen that router model, but if you want to make sure it's crippled you can try with a computer with linux on it and using the official tools to send a WPS pin and check whether it works:
https://www.wifi-libre.com/topic-924-metodo-universal-sin-modo-monitor-para-ataque-pixie-dust.html
maybe you can boot your mac with a kali linux live CD to try.

rofl0r, 2018-07-02:
btw i just found that our friend @kcdtv released my script automating those steps too:
https://www.wifi-libre.com/topic-924-metodo-universal-sin-modo-monitor-para-ataque-pixie-dust.html#p7989
be sure to install `pixiewps` prior to using it though

Tommrodrigues, 2018-07-03:
Thank you for the link, I've had a read through it and it looks very interesting. I'll have a fiddle around with the scripts when I'm next home.  

As for booting into Kali Linux, I've tried many times from a bootable USB image and it starts up perfectly; however, the problem is that neither the trackpad, keyboard, mouse or bluetooth work due to driver issues any many other people had had the same problem. I'm able to use a USB mouse but that is still a problem. (As for just using a virtual machine, everything works perfectly except for the wifi card as it appears as an ethernet connection rather than what it is supposed to)

Thanks once again for all your help so far though!

soxrok2212, 2018-07-04:
I know Realtek also released some patches to fix it and (fortunately) it seems that they did it right this time.

closed 2018-07-05 by kcdtv

hadilalash07, 2021-11-30:
#347 i have the same problem what you do to solved that

hadilalash07, 2021-11-30:
Hi my reaver tool at linux have eror timeout and receive death massage like you do you solve that problem