t6x/reaver-wps-fork-t6x #254: Wash - Json encryption data

DJDan, 2018-07-06:
Could the following:
"ENC" encryption type (WPA / WPA / OPN)
"CIPHER" (CCMP)
 "AUTH" (PSK)
Please be added to the wash -j Json output for all networks.
(Including wash -a all open devices)

Similar to what airodump-ng outputs.

Thanks

kcdtv, 2018-07-28:
The `-j` option is here to show the content of the probe as they are and will not be modified.

kcdtv, 2018-07-28:
You  said (not me) that wash with -j stdout "will never change in order to ensure compatibility with third party script" and that why i answered that
Than this information it is not in the "extend WPS parameter".
` -j, --json                           print extended WPS info as json`   
I don't realy see the advantage to know if it is wpa2 or wpa1, that doesn't change anything for WPS crack.. And an open network is open, if it has WPS enabled and you crack it you will get a rubbish key that is not in use. I could eventually understand it without -j but that would make another change in stdout and it is not a dta interisitng for WPS cracking. the nidce feature of wash without -j is that it is fast and straight to the  point and it holds in the usual 80 long size from the shell.  If you want airodump-ng style... why not use airodump-ng?

rofl0r, 2018-07-28:
yeah, the idea was to add it only in json output, not in 80-column normal mode (and eventually only with `-a`)
and yeah, i said `we won't change the format of the json output, except for adding new fields` ( e.g. https://github.com/t6x/reaver-wps-fork-t6x/issues/243#issuecomment-396583038 ) - and adding a field for encryption method would be just that.

kcdtv, 2018-07-30:
Hi there! 
Got it! Fine for me if you do so  :smiley_cat: