t6x/reaver-wps-fork-t6x #255: Reaver - Android compile error

DJDan, 2018-07-12:
Trying to compile Reaver using my Android LG G3 D855 that has LineageOS.
Linux localhost 3.4.113-Lineage-ga7e8977 #1 SMP PREEMPT Wed Jun 13 17:39:39 UTC 2018 armv71 Android.

I was hoping that I would be able to compile wash/reaver latest source on Android and then use nexmon drivers https://github.com/seemoo-lab/nexmon to turn my Android phone Broadcom wireless card into monitor mode (or use an Android supported external USB Wifi card).

However the Wash/Reaver source code did not seem to compile successfully at all.

===

I installed Termux android app. I then installed make, clang (6.0), libpcap, libpcap-dev.
./configure seemed to create a config.status and config.mak

However, when trying to do a 'make' I get the following error.

wpsmon.o: In function 'monitor':
/data/data/com/termux/files/home/reaver-wps-fork-t6x/src/wpsmon.c:331: undefined reference to 'ualarm'
wpsmon.o: In function 'parse_wps_settings':
clang-6.0: error: linker command failed with exit code 1 (ue -v to see invocation)
make: *** [ Makefile:123: reaver] Error 1

====

According to https://stackoverflow.com/questions/23447616/ualarm-equivalent-android-bionic-libc
it says the following:

ualarm() is not implemented in Android version of libc, bionic (check Bionic unistd.h). Moreover, ualarm() is obsolete.

The answer is in the Linux ualarm() man page: This function is obsolete. Use setitimer(2) or POSIX interval timers (timer_create(2), etc.) instead. – fadden May 3 '14 at 18:58

====

Can we change wpsmon.c:331 to use timer_create instead of ualarm by default or something?
or if the OS in android/does not support ualarm? or is there a better way to compile it?

Thanks

rofl0r, 2018-07-13:
i see limited usefulness in having wash/reaver on android, but if anyone wants to give it a try i'd be willing to merge a PR with proper documentation. Note that this is probably not the only issue. i'd expect other issues will pop up after this is "fixed". you can try to just comment out that line and see what happens when going on.

drygdryg, 2021-02-05:
I tried to comment out 2 lines with `ualarm` calls in the `wpsmon.c` file, and this allowed me to successfully build and run the project. I will try to replace obsolete `ualarm` with alternatives.

kimocoder, 2022-04-16:
Strange, I've been using Reaver on Android for years 👍 and it works well

rofl0r, 2022-04-16:
can we all please stop bumping year-old zombie threads :/
otherwise i'll have to lock discussion on every single on of them to collaborators.
anyway since you confirm it works, we can close this.

closed 2022-04-16 by rofl0r