t6x/reaver-wps-fork-t6x #259: reaver on macos: EAPOL request timeout occurred

2533245542, 2018-08-11:
Hi, I am using macbook pro 2017 and trying out reaver's WPS function, yet the EAPOL requests timed out every time.

I installed reaver with brew, used wash to find a WPS enabled router, and then I set up the monitor mode, and used reaver. Here are the codes I used.

This is the router information given by wash:
```
BSSID                  Channel       RSSI       WPS Version       WPS Locked        ESSID
XX:XX:XX:XX:XX:XX      11            -72        1.0               No                some_router
```

Then I enabled monitor mode by running:
`sudo ./airport en0 sniff 11`

Then I ran reaver in another terminal:
`sudo ./reaver -i en0 -b XX:XX:XX:XX:XX:XX -a -S -vv`

The output was:
Reaver v1.4 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

```
[+] Waiting for beacon from XX:XX:XX:XX:XX:XX
[+] Switching en0 to channel 11
[+] Associated with  XX:XX:XX:XX:XX:XX (ESSID: some_router)
[+] Trying pin 12345670
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
^C
[+] Nothing done, nothing to save.

```
`uname -a `
gives me this 
`Darwin MacBook-Pro-3.local 17.5.0 Darwin Kernel Version 17.5.0: Mon Mar  5 22:24:32 PST 2018; root:xnu-4570.51.1~1/RELEASE_X86_64 x86_64`

soxrok2212, 2018-08-11:
MacBooks/macOS have horrible monitor mode/injection support. Please try on a proper Linux system.

closed 2018-08-11 by soxrok2212

binarymaster, 2018-08-11:
> `sudo ./reaver -i en0 -b XX:XX:XX:XX:XX:XX -a -S -vv`

Why `-a` ?

kcdtv, 2018-08-11:
It seems that you are using **reaver 1.4** that was abandoned in... 2012. Or maybe **1.5.3**. 
Why did you erased the number version in your edits? That doesn't help.  
Anyway this versions are deprecated and we do not support them: We bring support from reaver 1.6 beta to the latest release (1.6.5)
Do not use your packet manager to install reaver. Use this repository. What is the point of giving mac support (and having an ugly bug with interfaces in some linux distros) if you don't use it? :smiley_cat: 
I guess you could use argument -N in your command line too. 

> Why -a ?

Does this option even existed in reaver 1.4? I can't remember. Anyway there is no option -a now.