t6x/reaver-wps-fork-t6x #265: RTL8812AU [X] ERROR: pcap_activate status -1

gantarone, 2018-11-21:
Raver  Error "RTL8812AU [X] ERROR: pcap_activate status -1" from ver. 1.6.4 to 1.6.5
(only ver form 1.4 to 1.6.3 work) 

err:
Reaver v1.6.5 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[X] ERROR: pcap_activate status -1
[X] PCAP: generic error code
couldn't get pcap handle, exiting

my wifi :
Bus 004 Device 010: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter

rofl0r, 2018-11-21:
what driver do you use for the 8812 ? what distro ? does it work when you put the card into monitor mode before starting reaver ?

gantarone, 2018-11-21:
Driver:
version:        v5.3.4_28226.20180614_COEX20171103-6c6d
Distro:
Linux kk 4.18.0-kali2-amd64 #1 SMP Debian 4.18.10-2kali1 (2018-10-09) x86_64 GNU/Linux

card in monitor mode work well !!!but when i use reaver ver  >= 1.6.4 I get : 
[X] ERROR: pcap_activate status -1

rofl0r, 2018-11-21:
try this https://github.com/t6x/reaver-wps-fork-t6x/issues/202#issuecomment-394125382

kcdtv, 2018-11-22:
Hi there!
I am back from hell! :smiley_cat: 
Before going to hell I noticed this issue too.
 It happens only with the latest version of the drivers: [v5.3.4](https://github.com/aircrack-ng/rtl8812au/tree/v5.3.4)
 It doesn't seems to be related #202 as there is no such issue with debian based systems 
  I have the chipset to do deeper testing...

rofl0r, 2018-11-22:
hi @kcdtv ! glad you're back.

in case you can reproduce this issue, could you test the potential fix i pointed out anyway?

kcdtv, 2018-11-22:
Hi again!
I've tried the fix you pointed out and it solved a part of the issue:
Wash now starts but remains"silent" (no AP shows up)  
[Fail](http://pix.toile-libre.org/upload/original/1542906852.png)
no packets received according to tcpdump

kcdtv, 2018-11-23:
> odd. if it affects tcpdump too, then reaver can't be blamed... 

It affects tcpdump **if i use it over wash** 
If i use tcpdump alone it works (in monitor mode). 
The rest of aircrack-ng suite works correctly too. 

>  if you feel adventurous, you could try to use git bisect on the driver's repo, to find the commit that introduced the bug.

  The problem is that the issue appears since the first publication of this version of the drivers ([v5.3.4](https://github.com/aircrack-ng/rtl8812au/tree/v5.3.4)). Reaver and wash never worked with them. 
  
  To solve the issue i also edited `pcap_set_promisc` to set the value of handle to 0 and wash worked.
In this case it is necessary to set  `pcap_set_rfmon` **and** `pcap_set_promisc` to **0** 
```
if (handle) {
		pcap_set_snaplen(handle, 65536);
		pcap_set_timeout(handle, 50);
		pcap_set_rfmon(handle, 0);
		pcap_set_promisc(handle, 0);
		if(!(status = pcap_activate(handle)))
			return handle;
		if(status == PCAP_ERROR_RFMON_NOTSUP) {
			pcap_set_rfmon(handle, 0);
			status = pcap_activate(handle);
			if(!status) return handle;
		}
```
In the link you gave i understood it was only pcap_set_rfmon, maybe you were speaking about bothes. 
Cheers :smiley_cat:

rofl0r, 2018-11-23:
thanks. i reported the issue here https://github.com/aircrack-ng/rtl8812au/issues/238 . let's wait a little to see what @kimocoder has to say...

kimocoder, 2018-11-29:
I have noticed this. Offline due to work offshore, brb!

kcdtv, 2018-12-10:
Thanks for the updating... The drivers 5.2.20 works pretty well for me and have already some interesting patches like this anti KRACK ones. And it works as one module for all chipset so it is very handy (well, for the weird people that have several devices with chipsets form this kind, it might not be so relevant for normal users)

> We've got together and we're trying to get in touch of Realtek about the matter on these issues/drivers or at least make an effort in getting it linux staging or something.

Good luck with that and thanks for your efforts  :smiley_cat: Maybe speak about it with Alfa Network, it might help to get in touch with the adequate people from realtek. If you go through regular channel of communication it is pretty hard to get somewhere... Cheers

kimocoder, 2019-02-11:
@kcdtv @rofl0r 

I've added some penetration testing abilities to the new Realtek (clean Realtek base, not from vendor this time) multichip driver from 2018, the rtl8188eus (with support for rtl8188eu and rtl8188etv too) and I was shocked, cause the base is v5.3.9 and actually got much better code quality and stability than prior. Anyhow, with this driver, Realtek actually added frame injection themselves (wtf?) and also got this "pcap_activate" status.. really annoying when I found this new base.

repo is [here](https://github.com/kimocoder/rtl8188eus)

rofl0r, 2019-02-11:
@kimocoder interesting. does the potential fix i suggested in https://github.com/t6x/reaver-wps-fork-t6x/issues/265#issuecomment-440806702 help ?

kimocoder, 2019-02-11:
I'll check tomorrow :+1: 
What's new since lwfinger's v5.2.2.4 is

* 802.11s MESH mode supporter
* Fixed signal leakage when testing MKK
* Fixed external LNA CCK RSSI bug

Som the injection is faster than the 8812au, packetloss (peaking) is fine and they've been alloher the place cleaning up, cause it's the same main base they always used for their multichip solution. Easy to pick patches/fixes from one to other.

I'll give you feedback, cause I need it resolved so the 8812/14/21 experience could jump up a notch at least. And also for those other chipsets.

kimocoder, 2019-02-15:
Ok. @rofl0r @kcdtv I may confirm turning "pcap_set_rfmon" and "pcap_set_promisc" to "0" works fine.
Diving into..

kcdtv, 2019-02-15:
Great news! I'll be around this weekend to check out and give you feedback. I have to try your latests commits in monitor mode. I tested some day ago hostapd in 5Ghz band with ac support and it worked smoothly. Reay cool. See you!

kimocoder, 2019-02-15:
Great, jumped over to v5.3.4 and fixed the injection just now, also lots of data wasn't dropped in txdesc film and urb was leaking a bit. But overall it should be better than the 5.2.20(2).

Will fiddle around and try find the cause of the pcap_activate status ☺️

ligmaDX800, 2019-03-22:
is the fix to this found yet ?
*help needed*

rofl0r, 2019-03-22:
@ligmaDX800 check PR #276

ligmaDX800, 2019-03-22:
rtl8188eu I have this one
tried your way still the same error

rofl0r, 2019-03-23:
> tried your way

you actually checked out the branch of that PR, compiled it, and still get the error ?

ligmaDX800, 2019-03-23:
I am actually new to linux so I dunno how it works correct me if I am wrong.
I copied the lines of code from commit that were changed and replaced them in the src/init.c file.

ligmaDX800, 2019-03-23:
*update*
its working thanks alot mate <3
I just read the read me file and it taught me how to do certain things I wasn't doing.
@rofl0r

rofl0r, 2019-03-23:
alright thanks for testing

ligmaDX800, 2019-03-23:
I have come across another issue after running the reaver command following happens:

reaver -b 98:DE:D0:B2:F6:CA -i wlan0 -c 10 -vv

Reaver v1.6.5-git-18-g48a0a8b WiFi Protected Setup Attack ToolCopyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
[+] Switching wlan0 to channel 10
[+] Waiting for beacon from 98:DE:D0:B2:F6:CA
[+] Received beacon from 98:DE:D0:B2:F6:CA
[+] Vendor: AtherosC
[+] Trying pin "12345670"
[+] Sending authentication request
[!] WARNING: Receive timeout occurred

rofl0r, 2019-03-23:
new issue -> [new issue]

but check https://github.com/t6x/reaver-wps-fork-t6x/wiki/Troubleshooting first

rofl0r, 2019-04-06:
ok since #276 was merged this can be closed.

closed 2019-04-06 by rofl0r

MichaelChay, 2019-05-03:
> To solve the issue i also edited `pcap_set_promisc` to set the value of handle to 0 and wash worked.
> In this case it is necessary to set `pcap_set_rfmon` **and** `pcap_set_promisc` to **0**
>
> ```
> if [...]

@kimocoder where to locate this file in directory for editing  `pcap_set_rfmon` **and** `pcap_set_promisc` to **0** ????

rofl0r, 2019-05-03:
> @kimocoder where to locate this file in directory for editing pcap_set_rfmon and pcap_set_promisc to 0 ????

grep for pcap_set_snaplen. `grep -r pcap_set_snaplen .`

MichaelChay, 2019-05-03:
TQ @rofl0r ...... :+1:

MichaelChay, 2019-05-03:
@rofl0r @kimocoder 
root@hostname:~/rtl8812au# **grep -r pcap_set_snaplen**
root@hostname:~/rtl8812au#

which file to edit and the file location ???

MichaelChay, 2019-05-03:
no result

MichaelChay, 2019-05-03:
grep -r pcap_set_snaplen .
it take too long to search in the root directory !!!!!!

do you have the file path ????

rofl0r, 2019-05-03:
> it take too long to search in the root directory !!!!!!

try it inside the source directory of reaver

NeilKerman, 2019-05-25:
> > it take too long to search in the root directory !!!!!!
> 
> try it inside the source directory of reaver

Where do i find the source directory of reaver? I am new to linux so i really dont know how to find it. I also tried using grep, but it doesn't gives a proper output.

NeilKerman, 2019-05-25:
> _update_
> its working thanks alot mate <3
> I just read the read me file and it taught me how to do certain things I wasn't doing.
> @rofl0r

May you please tell me what are the steps you did to fix it?

iGriefU, 2019-06-05:
> > odd. if it affects tcpdump too, then reaver can't be blamed...
> 
> It affects tcpdump **if i use it over wash**
> If i use tcpdump alone it works (in monitor mode).
> The rest of aircrack-ng suite works correctly too.
> I am not blaming reaver; There is definitely something new with this version of the drivers as there is no problem with others branches. As this drivers are unofficial and a bit "hackish" we can expect some strange little things.
> 
> > if you feel adventurous, you could try to use git bisect on the driver's repo, to find the commit that introduced the bug.
> 
> The problem is that the issue appears since the first publication of this version of the drivers ([v5.3.4](https://github.com/aircrack-ng/rtl8812au/tree/v5.3.4)). Reaver and wash never worked with them.
> 
> To solve the issue i also edited `pcap_set_promisc` to set the value of handle to 0 and wash worked.
> In this case it is necessary to set `pcap_set_rfmon` **and** `pcap_set_promisc` to **0**
> 
> ```
> if (handle) {
> 		pcap_set_snaplen(handle, 65536);
> 		pcap_set_timeout(handle, 50);
> 		pcap_set_rfmon(handle, 0);
> 		pcap_set_promisc(handle, 0);
> 		if(!(status = pcap_activate(handle)))
> 			return handle;
> 		if(status == PCAP_ERROR_RFMON_NOTSUP) {
> 			pcap_set_rfmon(handle, 0);
> 			status = pcap_activate(handle);
> 			if(!status) return handle;
> 		}
> ```
> 
> In the link you gave i understood it was only pcap_set_rfmon, maybe you were speaking about bothes.
> Cheers 😺

**Where is the destination of the mentioned file that should be edited ?** 

My output  from "reaver -i wlan0 -b _MAC_ is :
[X] ERROR: pcap_activate status -1
[X] PCAP: generic error code
couldn't get pcap handle, exiting

rofl0r, 2019-06-05:
there's no need to edit anything, the patches mentioned here are already merged into master.
if you have problems with pcap, you should probably first try to get aircrack-ng working (which i assume doesn't work either, because you seem to have forgotten to activate monitor mode...).
if you have further issues, open a new issue. i will close this now for good since the original issue is long resolved.