t6x/reaver-wps-fork-t6x #270: Fail send message 2 and do not receive message 3, received nack

darkalexpp, 2019-02-06:
Reaver 1.6.5 + pixie-dust 1.4, libpcap 1.7.4

I have problems getting the M3, get the M1, build the M2, send the M2, but I do not receive M3, I receive NACK.
E tried from other platforms, and if I manage to receive M3, I have tried to perform an attack without -K (without pixie dust) and if I receive M3

[+] Trying pin ********.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received WSC NACK
[+] Trying pin ********.
....

soxrok2212, 2019-02-06:
WPS is locked out. Reboot the router.

darkalexpp, 2019-02-06:
The router is not locked. When the command WASH is executed. It does not appear locked.

When I try the attack with reaver 1.4 (the one that does not have pixie dust) it executes it correctly until the m7

soxrok2212, 2019-02-06:
Try executing first with '-N' and if that doesn't work, try with '-F'.

kcdtv, 2019-02-06:
If you receive a NACK after sending M2 it very likely means that WPS isn't properly activated in PIN mode.
What model of router are you testing? 
Did you try using the correct PIN with -p option? 
Show us please the WPS settings from your routeur interface.

darkalexpp, 2019-02-06:
the WPS is enabled on the router and is not blocked, and tested with other versions of reaver 1.4 that does not include pixie dust, and works well.
Try reaver 1.6.5 from another device and it works fine.
but in this device (bcm4335) it only reaches M2 and then NACK, with and without -K
Viewing the data with wireshark, I see that the router sends several M1s to the device before the device sends its M2.
It seems that it sends me NACK because the M2 is incorrect, so it opens multiple different sections.
How can I correct this?

kcdtv, 2019-02-06:
i did not see that you had success with reaver 1.4
As suggested before use option `-N `with full verbose (`-vvv` with **3** v) and copy paste the stdout here. 
Correcting the M2 on the fly is very tricky and it would be anyway much better to fix this.
So we need to have a full stdout with `-vvv` to get as much information as possible.

darkalexpp, 2019-02-06:
Reaver 1.6.5 + pixie-dust 1.4, libpcap 1.7.4
wireless: BCM4335  MAC: 40:f3:08:4d:85:f3

wash -i wlan0  
  
BSSID               Ch  dBm  WPS  Lck  Vendor    ESSID
--------------------------------------------------------------------------------
D8:FE:E3:82:42:08    9  -41  2.0  No   RealtekS  Conejos Patios

reaver -i wlan0 -b D8:FE:E3:82:42:08 -c 9 -K -N -vvv -F  

Reaver v1.6.5-git-12-gf936d3d WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner

[+] Switching wlan0 to channel 9
[+] Waiting for beacon from D8:FE:E3:82:42:08
[+] Received beacon from D8:FE:E3:82:42:08
[+] Vendor: RealtekS
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
     31 32 33 34 35 36 37 30                           12345670        
WPS: Selected registrar information changed
WPS: Internal Registrar selected (pbc=0)
WPS: sel_reg_union
WPS: set_ie
WPS: cb_set_sel_reg
WPS: Enter wps_cg_set_sel_reg
WPS: Leave wps_cg_set_sel_reg early
WPS: return from wps_selected_registrar_changed
[+] Trying pin "12345670"
send_packet called from deauthenticate() //home/alex/Escritorio/Krack/reaver-wps-android/src/android_external_libpcap/../80211.c:337
send_packet called from authenticate() //home/alex/Escritorio/Krack/reaver-wps-android/src/android_external_libpcap/../80211.c:368
[+] Sending authentication request
send_packet called from associate() //home/alex/Escritorio/Krack/reaver-wps-android/src/android_external_libpcap/../80211.c:421
[+] Sending association request
[+] Associated with D8:FE:E3:82:42:08 (ESSID: Conejos Patios)
[+] Sending EAPOL START request
send_packet called from send_eapol_start() //home/alex/Escritorio/Krack/reaver-wps-android/src/android_external_libpcap/../send.c:48
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() //home/alex/Escritorio/Krack/reaver-wps-android/src/android_external_libpcap/../send.c:81
send_packet called from resend_last_packet() //home/alex/Escritorio/Krack/reaver-wps-android/src/android_external_libpcap/../send.c:161
send_packet called from resend_last_packet() //home/alex/Escritorio/Krack/reaver-wps-android/src/android_external_libpcap/../send.c:161
WPS: Processing received message (len=417 op_code=4)
WPS: Received WSC_MSG
WPS: Unsupported attribute type 0x1049 len=6
WPS: Parsed WSC_MSG
WPS: Received M1
WPS: UUID-E - hexdump(len=16): 63 04 12 53 10 19 20 06 12 28 d8 fe e3 82 42 08
WPS: Enrollee MAC Address d8:fe:e3:82:42:08
WPS: Enrollee Nonce - hexdump(len=16): 45 a8 18 95 53 b8 82 70 18 a3 6f 6b 3d 66 95 d3
WPS: Enrollee Authentication Type flags 0x21
WPS: No match in supported authentication types (own 0x0 Enrollee 0x21)
WPS: Workaround - assume Enrollee does not advertise supported authentication types correctly
WPS: Enrollee Encryption Type flags 0x9
WPS: No match in supported encryption types (own 0x0 Enrollee 0x9)
WPS: Workaround - assume Enrollee does not advertise supported encryption types correctly
WPS: Enrollee Connection Type flags 0x1
WPS: Enrollee Config Methods 0x2688 [Display] [PBC]
WPS: Enrollee Wi-Fi Protected Setup State 2
WPS: Manufacturer - hexdump_ascii(len=18):
     44 2d 4c 69 6e 6b 20 43 6f 72 70 6f 72 61 74 69   D-Link Corporati
     6f 6e                                             on              
WPS: Model Name - hexdump_ascii(len=13):
     44 2d 4c 69 6e 6b 20 52 6f 75 74 65 72            D-Link Router   
WPS: Model Number - hexdump_ascii(len=8):
     44 49 52 2d 39 30 35 4c                           DIR-905L        
WPS: Serial Number - hexdump_ascii(len=13):
     32 30 30 37 30 34 31 33 2d 30 30 30 31            20070413-0001   
[... 87 lines cut ...]
[+] Waiting for beacon from D8:FE:E3:82:42:08
DEBUG external/reaver-wps/80211.c (229): Red AP beacon
DEBUG external/reaver-wps/80211.c (235): deauthenticate() done
DEBUG external/reaver-wps/80211.c (241): authenticate() done
DEBUG external/reaver-wps/80211.c (245): end while associate_recv_loop()
DEBUG external/reaver-wps/80211.c (250): tries < MAX_AUTH_TRIES
DEBUG external/reaver-wps/80211.c (256): associate() done in loop
[+] Associated with D8:FE:E3:82:42:08 (ESSID: Conejos Patios)
[+] Trying pin 12345670
DEBUG external/reaver-wps/80211.c (229): Red AP beacon
DEBUG external/reaver-wps/80211.c (235): deauthenticate() done
DEBUG external/reaver-wps/80211.c (241): authenticate() done
DEBUG external/reaver-wps/80211.c (245): end while associate_recv_loop()
DEBUG external/reaver-wps/80211.c (250): tries < MAX_AUTH_TRIES
DEBUG external/reaver-wps/80211.c (256): associate() done in loop
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Trying pin 00005678

rofl0r, 2019-02-06:
thanks for the detailed output. would you mind checking a few of the other releases from here: https://github.com/t6x/reaver-wps-fork-t6x/releases to find out which one is the first "bad" release ?

darkalexpp, 2019-02-07:
Hello everyone, I was finally able to solve the problem of the badly generated M2 message, everything was a bad configuration that occurred at the time of compiling, in the file crypto/crypto_internal-modexp.c specifically by the file bignum.c
the data was being truncated.

kcdtv, 2019-02-07:
Thanks for your feedback
Did you use `./configure` before compiling?

darkalexpp, 2019-02-07:
Yes, I use ./configure before compiling.

darkalexpp, 2019-02-07:
Now I have another problem, I already received the M3 but sometimes it executes pixie dust, and other times it passes and sends M4 to the router, and no pixie dust is activated, which leads to the router being locked.
this only happens in reaver 1.6.5, probe with 1.6b and it works correctly,
throughout the day I will try other versions

soxrok2212, 2019-02-07:
Please open a new issue for this.

closed 2019-02-07 by soxrok2212

Mostapha2005, 2023-09-04:
please i need a solution , in the stat i was using waicut without any problems but now when im starting the pixie dust , it,s just stopping at M1 and showing me *THE RPC SERVER IS UNAVAILABLE*

rofl0r, 2023-09-05:
this is not a waicut help forum, bring it there.