t6x/reaver-wps-fork-t6x #272: reaver stops guessing pins when use with -K

logan001, 2019-02-23:
I just notice that when i try to use reaver with -K option reaver will stop with one attempt and that;s it.

Testing on the router reaver with a pin (not the router pin) seems to make pixie discover the correct pin
Changing the pin provided to reaver to another random one a second time resulted in failure for pixie.

Is there any option to make reaver behave like without -K but to also use pixiewps after each pin test so that reaver can keep guessing the pin but also to pass the corresponding variables to pixie

rofl0r, 2019-02-23:
pixiedust attacks either works always, or never (with a specific router). so there's no reason to keep testing the attack over and over.

closed 2019-02-23 by rofl0r

logan001, 2019-02-23:
if that is so than why the variable are different with the same pin?

rofl0r, 2019-02-23:
which variable ?

logan001, 2019-02-23:
s/z/a/n/r

rofl0r, 2019-02-23:
those are cryptographic nonces. they're supposed to be different each time...

logan001, 2019-02-23:
seems that i can use  reaver-wps-fork-t6x/tools/logfilter.py and -vvv output

logan001, 2019-02-23:
strange. why would pixie need them if they are suppose to change. and why would pixie will not work with different set of variables?
well i'm not into programing/wifi protocols/whatever.
for me this seems strange.

kimocoder, 2019-02-23:
WPS pin attack is one type of attack, pixiewps is another attack vector again..

rofl0r, 2019-02-23:
see the README and the wiki on this repo. there are articles and links to articles which explain in great detail how these attacks works.