t6x/reaver-wps-fork-t6x #275: Constant deauth in MacOS

freehalalmeats, 2019-03-16:
When attempting to use Reaver on macOS:

sudo reaver -i en0 -b 14:A5:1A:09:7F:A3 -F -vvv

Reaver v1.6.5-git-18-g48a0a8b WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Waiting for beacon from 14:A5:1A:09:7F:A3
[+] Switching en0 to channel 1
[+] Switching en0 to channel 2
[+] Switching en0 to channel 3
[+] Switching en0 to channel 4
[+] Switching en0 to channel 5
[+] Received beacon from 14:A5:1A:09:7F:A3
[+] Vendor: Broadcom
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
     31 32 33 34 35 36 37 30                           12345670        
WPS: Selected registrar information changed
WPS: Internal Registrar selected (pbc=0)
WPS: sel_reg_union
WPS: set_ie
WPS: cb_set_sel_reg
WPS: Enter wps_cg_set_sel_reg
WPS: Leave wps_cg_set_sel_reg early
WPS: return from wps_selected_registrar_changed
[+] Trying pin "12345670"
send_packet called from deauthenticate() 80211.c:337
send_packet called from authenticate() 80211.c:368
[+] Sending authentication request
send_packet called from associate() 80211.c:421
[+] Sending association request
[+] Associated with 14:A5:1A:09:7F:A3 (ESSID: --Hepburn--)
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
[+] Received deauth request
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
[+] Received deauth request
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161

I have tried moving closer to the router, too.

Are there any other mac compatible pieces of software that I can use to handle authentication? (aireplay-ng isn't supported, it seems)

rofl0r, 2019-03-16:
```
[+] Received deauth request
```
some routers do that, i haven't yet figured out why. maybe try with another type?

freehalalmeats, 2019-03-17:
Have tried with several routers now, but the results are all the same. Most of the routers are running WPS version 2.0. Could this possibly be the cause?

rofl0r, 2019-03-17:
no, that's not related. maybe the mac driver adds some invalid stuff to the packets, causing the routers to force a deauth. apparently mac monitor drivers are pretty buggy, so far there's only one apple device known working, iirc macbook white (2004) which @djdan owns

elig0n, 2020-07-27:
I get the same type of response on linux but without 
`[+] Received deauth request`
Observing the packets it seems the AP sends 4 Deauthentication packets after each `EAP Start` stating the reason:
`Reason code: Class 2 frame received from nonauthenticated STA (0x0006)`
but also there's Code 4 Failure `EAP packet`s (type 0) after a while sent together right after `Start`

rofl0r, 2020-07-27:
> I have also tried sending aireplay-ng fake association before running reaver to no avail.
> My antenna is less than 1 meter from the AP and I've also tried extending timeouts durations.

could you sniff with a second antenna when you associate to the AP using wpa_supplicant or anything that works, and compare what they do differently ?