t6x/reaver-wps-fork-t6x #277: Reaver WARNING: receive timeout

ligmaDX800, 2019-03-25:
Following error occurs while running reaver:

reaver -b 98:DE:D0:B2:F6:CA -i wlan0 -c 10 -vv
Reaver v1.6.5-git-18-g48a0a8b WiFi Protected Setup Attack ToolCopyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
[+] Switching wlan0 to channel 10
[+] Waiting for beacon from 98:DE:D0:B2:F6:CA
[+] Received beacon from 98:DE:D0:B2:F6:CA
[+] Vendor: AtherosC
[+] Trying pin "12345670"
[+] Sending authentication request
[!] WARNING: Receive timeout occurred
[+] Sending authentication request
[!] WARNING: Receive timeout occurred
[+] Sending authentication request

rofl0r, 2019-03-25:
did you check the article in the wiki i linked twice? report what happens when you follow the steps outlined there.

ligmaDX800, 2019-03-27:
Tried troubleshooting, still same issue.

rofl0r, 2019-03-27:
> report what happens when you follow the steps outlined there

since you apparently are incapable of following simple instructions, i don't feel like i would want to invest more time trying to help you.

closed 2019-03-27 by rofl0r

ligmaDX800, 2019-03-28:
Well I wasn't sure of what you meant English is not my first language, forgive me for my incompetence. I will re troubleshoot it and report each step. Hope you reconsider helping me.