t6x/reaver-wps-fork-t6x #278: How Unistall this version?

Veigalex, 2019-04-14:
I don unistall this version.
Please help

apt-get unistal reaver not found...

closed 2019-04-14 by soxrok2212

Veigalex, 2019-04-14:
I can not uninstall!
Please help me!

Veigalex, 2019-12-28:
Please

Asux30, 2019-12-28:
> I don unistall this version.
> Please help
> 
> apt-get unistal reaver not found...

apt-get **remove** reaver

kcdtv, 2019-12-28:
For information `apt-get` is deprecated and `remove` does not uninstall the unnecessary dependencies.
And don't forget to get root privilege to execute the command  
`sudo apt autoremove reaver`
Saying so if you use the packet manager form your distribution it has nothing to do with our github repository, your issue is completely "off topic" here. 
Please check your distribution documentation and search the internet before opening issues, a 10 seconds search would have been enough to answer this question that you asked... more than 10 months ago...