t6x/reaver-wps-fork-t6x #279: Constant timeout on rtl8188eu (Raspberry Pi 3 B+ ARM64)

sh1ku, 2019-04-27:
I'm unable to get reaver to work with my network card, a TP-WN722N V3. Aireplay-ng association and inject test works fine, but reaver gets stuck in a loop at "Sending authentication request", "WARNING: Receive timeout occurred". It's working fine using the on-board wifi chipset. I'm attaching Reaver's output, as well as airmon-ng's output and a test cap I got with tcpdump following the Troubleshooting wiki page.

[debug.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/3123546/debug.zip)

rofl0r, 2019-04-27:
i don't see a response from the router. can i have a pcap from aircrack-ng authentication/association for comparison ?

sh1ku, 2019-04-27:
Sure.

cap2_injtest.cap includes output for this:
> aireplay-ng --test wlan0

and cap2_assoctest.cat includes output for this:
>aireplay-ng -1 0 -a E2:41:36:89:F1:78 -e "Wifi_Alberto" wlan0

[captest.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/3124284/captest.zip)

rofl0r, 2019-04-27:
there are 3 differences:
1) aircrack starts with sequence number 0, we with 1. i just pushed a "fix" changing that.
please pull, rebuild (make clean; make) and try again.
2) we send a deauth packet, before sending the auth packet.
if the fix from 1 doesnt help, try commenting out this line in 80211.c
```diff
                         case 0:
-                                deauthenticate();
+                                // deauthenticate();
                                 state++;
                                 break;
```
3) aircrack uses duration 314, whereas we use duration 52.
if the fixes from 1 and 2 don't help. try changing this line in builder.c
```diff
         static uint16_t frag_seq;
   
-        fh->duration = end_htole16(52); /* DEFAULT_DURATION */
+        fh->duration = end_htole16(314); /* DEFAULT_DURATION */
         fh->fc = end_htole16(fc);
         fh->frag_seq = end_htole16(frag_seq);
```
lastly i noticed that a different MAC addr connects to the AP in both pcaps. i assume you were using the same adapter but changed the MAC address. if not, tell me.

sh1ku, 2019-04-27:
Same result. I think I did the changes right, as there is no compilation error and deauthenticate() no longer seems to be called. I included reaver's output, anyway, just in case, and a tcpdump cap.

The network card is the same one.

[debug2.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/3124371/debug2.zip)

rofl0r, 2019-04-27:
thanks. there's another change in aircrack regarding the radiotap header.
please stash your local changes (`git stash`) and pull what i just pushed.

rebuild with make clean, then try whether it fixes the issue.
if it doesn't, use `git stash pop` to restore your changes and try again.
if that still doesn't help, make the following change to defs.h:
```diff
-  
+ #define RADIOTAP_HEADER_WITH_RATE
 #pragma pack(1)
 struct radio_tap_header
 {       
         uint8_t revision; 
         uint8_t pad;
         le16 len;  
         le32 flags;
 #ifdef RADIOTAP_HEADER_WITH_RATE
```
and again, `make clean`, and rebuild

rofl0r, 2019-04-27:
btw @kimocoder do you think the RTL8818EU driver could be fixed to ignore the radiotap header as all (or most) other drivers do?

sh1ku, 2019-04-28:
Changing the defs.h to define RADIOTAP_HEADER_WITH_RATE made the trick. I used the version with all the other changes (80211.c, builder.c). 

If you want me to test something else, please tell me.

Thanks for the help.

rofl0r, 2019-04-28:
cool, thanks for testing. would you mind undoing the other 2 changes again with git stash and tell me whether it makes a difference ?

rofl0r, 2019-04-28:
oh, and please tell me which driver you use. i have the suspicion the driver has a bug and assumes a size of 12 for the radio tap header, since that's what aircrack uses...

sh1ku, 2019-04-28:
For the driver, I use kimocoder's one (I believe it's a fork of the aircrack-ng one?). [This one](https://github.com/kimocoder/rtl8188eus)

Reaver works apparently the same without the 2 changes mentioned.

rofl0r, 2019-04-28:
great. i found the bug and filed a PR. would you mind testing it ?
https://github.com/kimocoder/rtl8188eus/pull/10

sh1ku, 2019-04-28:
Got the latest driver version. Seems to fix the issue, as a fresh reaver compilation works fine, without having to edit anything.

closed 2019-04-28 by rofl0r