t6x/reaver-wps-fork-t6x #281: Using make on Debian fails due to missing -lpcap -lm

mzpqnxow, 2019-05-03:
make fails when linking because it can't find symbols in libm and libpcap. Adding -lpcap and -lm fixes the issue..

mzpqnxow, 2019-05-03:
```
cc -DCONF_DIR='""'  -Wno-unused-variable -Wno-unused-function -Wno-pointer-sign -Ilibwps -I. -Ilwe -DCONFIG_IPV6 -Ilibwps -I. -Ilwe globule.o init.o sigint.o iface.o sigalrm.o misc.o session.o send.o pins.o 80211.o builder.o keys.o crc.o pixie.o version.o pcapfile.o exchange.o argsparser.o wpscrack.o wpsmon.o cracker.o main.o libwps/libwps.o wps/wps_attr_build.o wps/wps_attr_parse.o wps/wps_attr_process.o wps/wps.o wps/wps_common.o wps/wps_dev_attr.o wps/wps_enrollee.o wps/wps_registrar.o wps/wps_ufd.o utils/base64.o utils/common.o utils/ip_addr.o utils/radiotap.o utils/trace.o utils/uuid.o utils/wpa_debug.o utils/wpabuf.o utils/os_unix.o utils/vendor.o utils/eloop.o tls/asn1.o tls/bignum.o tls/pkcs1.o tls/pkcs5.o tls/pkcs8.o tls/rsa.o tls/tlsv1_client.o tls/tlsv1_client_read.o tls/tlsv1_client_write.o tls/tlsv1_common.o tls/tlsv1_cred.o tls/tlsv1_record.o tls/tlsv1_server.o tls/tlsv1_server_read.o tls/tlsv1_server_write.o tls/x509v3.o crypto/aes-cbc.o crypto/aes-ctr.o crypto/aes-eax.o crypto/aes-encblock.o crypto/aes-internal.o crypto/aes-internal-dec.o crypto/aes-internal-enc.o crypto/aes-omac1.o crypto/aes-unwrap.o crypto/aes-wrap.o crypto/des-internal.o crypto/dh_group5.o crypto/dh_groups.o crypto/md4-internal.o crypto/md5.o crypto/md5-internal.o crypto/milenage.o crypto/ms_funcs.o crypto/rc4.o crypto/sha1.o crypto/sha1-internal.o crypto/sha1-pbkdf2.o crypto/sha1-tlsprf.o crypto/sha1-tprf.o crypto/sha256.o crypto/sha256-internal.o crypto/crypto_internal.o crypto/crypto_internal-cipher.o crypto/crypto_internal-modexp.o crypto/crypto_internal-rsa.o crypto/tls_internal.o crypto/fips_prf_internal.o lwe/iwlib.o  -o reaver
globule.o: In function `globule_deinit':
globule.c:(.text+0x181): undefined reference to `pcap_close'
init.o: In function `capture_init':
init.c:(.text+0x2c0): undefined reference to `pcap_open_offline'
init.c:(.text+0x2ed): undefined reference to `pcap_create'
init.c:(.text+0x30d): undefined reference to `pcap_set_snaplen'
init.c:(.text+0x31e): undefined reference to `pcap_set_timeout'
init.c:(.text+0x32f): undefined reference to `pcap_set_rfmon'
init.c:(.text+0x340): undefined reference to `pcap_set_promisc'
init.c:(.text+0x34c): undefined reference to `pcap_activate'
init.c:(.text+0x375): undefined reference to `pcap_set_rfmon'
init.c:(.text+0x381): undefined reference to `pcap_activate'
init.c:(.text+0x3f9): undefined reference to `pcap_close'
misc.o: In function `pcap_sleep':
misc.c:(.text+0x33b): undefined reference to `pcap_close'
send.o: In function `send_packet_real':
send.c:(.text+0x38f): undefined reference to `pcap_inject'
80211.o: In function `next_packet':
80211.c:(.text+0xd3): undefined reference to `pcap_next_ex'
80211.o: In function `has_rt_header':
80211.c:(.text+0x17cb): undefined reference to `pcap_datalink'
wpsmon.o: In function `sigint_handler':
wpsmon.c:(.text+0x413): undefined reference to `pcap_breakloop'
wpsmon.o: In function `wash_main':
wpsmon.c:(.text+0x8bf): undefined reference to `pcap_compile'
wpsmon.c:(.text+0x8eb): undefined reference to `pcap_geterr'
wpsmon.c:(.text+0x922): undefined reference to `pcap_setfilter'
cracker.o: In function `crack':
cracker.c:(.text+0x56c): undefined reference to `pcap_close'
lwe/iwlib.o: In function `iw_float2freq':
iwlib.c:(.text+0x141e): undefined reference to `log10'
iwlib.c:(.text+0x1423): undefined reference to `floor'
iwlib.c:(.text+0x146c): undefined reference to `pow'
iwlib.c:(.text+0x147e): undefined reference to `floor'
lwe/iwlib.o: In function `iw_freq2float':
iwlib.c:(.text+0x150d): undefined reference to `pow'
lwe/iwlib.o: In function `iw_dbm2mwatt':
iwlib.c:(.text+0x1958): undefined reference to `pow'
iwlib.c:(.text+0x195d): undefined reference to `floor'
lwe/iwlib.o: In function `iw_mwatt2dbm':
iwlib.c:(.text+0x197c): undefined reference to `log10'
iwlib.c:(.text+0x1991): undefined reference to `ceil'
collect2: error: ld returned 1 exit status
Makefile:123: recipe for target 'reaver' failed
make: *** [reaver] Error 1
debian% 
```

mzpqnxow, 2019-05-03:
aaaannnddd I just realized the build instructions are out of date and there is now a `./configure` that probably deals with this.. closing

closed 2019-05-03 by mzpqnxow

rofl0r, 2019-05-04:
> I just realized the build instructions are out of date 

maybe yours. reaver always required a configure script. and it was always documented that way.