t6x/reaver-wps-fork-t6x #282: ERROR: pcap_activate status -8 and pcap_activate status -1

vav4eg, 2019-05-10:
The first problem was:

> wash -i wlan1
> [X] ERROR: pcap_activate status -1
[X] PCAP: generic error code
couldn't get pcap handle, exiting

Problems solved by comment https://github.com/t6x/reaver-wps-fork-t6x/issues/265#issuecomment-441272085

There is now a second problem

> wash -i wlan1
> [X] ERROR: pcap_activate status -8
> [X] PCAP: no permission to open the device
> couldn't get pcap handle, exiting

rofl0r, 2019-05-12:
are you root? if not check strace output

vav4eg, 2019-05-13:
The error has gone. But I have a big problem with the drivers on the adapter. My TP-Link Archer T4U AC1300 V3 adapter (RTL88x2BU), but to my great regret I can not find a driver in which the River would work. I tried different drivers on different systems that I found here (RTL88x2BU). The card is transferred to the monitoring mode only by the command - sudo iw dev wlan0 set type monitor. I will be very grateful if someone advises a suitable driver for my card or looks at the drivers I tried for the river. I am very frustrated by the fact that manufacturers do not want to support the Linux system and have to ask for help and simple programmers on this site.

Drivers that I used for the river.
https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959
https://github.com/RinCat/RTL88x2BU-Linux-Driver

kimocoder, 2019-05-26:
@kcdtv the issue is showing itself most newer Realtek releases it seems.
btw, take a look at rtl8812au v5.6.4 (8814 and 8821 not supported yet), but this driver works really good, finally!

Also, take a look at https://lwn.net/Articles/768372/
Finally, their moving over to mac80211, I bet i could get use of that release when it's finished.

vav4eg, 2019-06-03:
> @kcdtv the issue is showing itself most newer Realtek releases it seems.
> btw, take a look at rtl8812au v5.6.4 (8814 and 8821 not supported yet), but this driver works really good, finally!
>
> Also, [...]

Unfortunately, the driver proposed by you did not fit. The device is not detected.

kimocoder, 2019-06-15:
More reading on the issue
https://github.com/secdev/scapy/issues/2076

Rueles97, 2020-07-19:
> The first problem was:
> 
> > wash -i wlan1
> > [X] ERROR: pcap_activate status -1
> > [X] PCAP: generic error code
> > couldn't get pcap handle, exiting
> 
> Problems solved by comment [#265 (comment)](https://github.com/t6x/reaver-wps-fork-t6x/issues/265#issuecomment-441272085)
> 
> There is now a second problem
> 
> > wash -i wlan1
> > [X] ERROR: pcap_activate status -8
> > [X] PCAP: no permission to open the device
> > couldn't get pcap handle, exiting

did yall find a soluision for this help pls

vav4eg, 2020-07-19:
> did yall find a soluision for this help pls

Not

VINAYNIC, 2020-10-10:
> Not

Any new updates that fixes pcap_activate 
For ​ rtl8188eu

kcdtv, 2020-11-03:
First don't forget to run the command with "root privileges" (use `sudo` in debian based systems) 
If you have the required privileges, run airodump-ng for one second, stop it , then run wash. 
It should work