t6x/reaver-wps-fork-t6x #288: wash: Abnormal output in terminal (OpenWRT)

adde88, 2019-09-19:
**Device**: Hak5 WiFi Pineapple NANO + TETRA
**Architecture**: MIPS, ar71xx
**OS**: Tested on OpenWRT 15.05, 18.06.0, and 19.07

**ISSUE**:
Wash works works perfectly on OpenWRT 15.05. But we're having issues on OpenWRT-18.06.0 up to 19.07.
When launching wash there's almost no output to terminal.
I've tried using every flag, and even --all does not help.
On a few occasions though there is some output displayed, but those MAC addresses should not be there, and they all have the ESSID: <null>
There also seems to be a pattern to the strange MAC addresses. The last two octets of the MAC ALWAYS seems to be zero. Example (50:B1:7E:10:00:00, 52:2F:6A:E6:00:00, 52:BE:6A:43:00:00)

Using wash to read a capture from airodump-ng works.

I've tried using the version of reaver on openwrt's package manager, as well as compiling MANY different commits of reaver. But the problem is there on everyone i've tried.

I've run an strace on wash if that helps solving the issue. I'm including the links here to make the post more readable.

The latest OpenWRT 19.07 is using gcc-7.4.0 and musl-1.1.23, while OpenWRT 15.05 was using gcc-4.8 and uClibc-0.9.33.2.

Output of "ldd /usr/bin/wash" on the working version of OpenWRT 15.05:
```
	libm.so.0 => /lib/libm.so.0 (0x77542000)
	libpcap.so.1 => /usr/lib/libpcap.so.1 (0x77504000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x774e0000)
	libc.so.0 => /lib/libc.so.0 (0x77473000)
	ld-uClibc.so.0 => /lib/ld-uClibc.so.0 (0x77568000)
```

Output of "ldd /usr/bin/wash" on latest OpenWRT 19.07
```
	/lib/ld-musl-mips-sf.so.1 (0x77e88000)
	libpcap.so.1 => /sd/usr/lib/libpcap.so.1 (0x77e4c000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77e28000)
	libc.so => /lib/ld-musl-mips-sf.so.1 (0x77e88000)
```

**LINKS TO STRACE OUTPUT**:

OpenWRT (19.07) 30 second strace: (NOT-WORKING)
https://pastebin.com/5PS0tg8A

OpenWRT (19.07) longer strace, with buggy MAC addresses and ESSIDs: (NOT WORKING)
https://pastebin.com/BFrKQznx

OpenWRT (18.06.0) 30 second strace: (NOT-WORKING)
https://pastebin.com/NvWYhaQD

OpenWRT (15.05) 30 second strace: (WORKING)
https://pastebin.com/e955Xctb

CONFIG.LOG (OpenWRT 19.07): 
https://pastebin.com/7rkWnUyu

CONFIG.LOG (OpenWRT 15.05):
https://pastebin.com/vuPBF26v

rofl0r, 2019-09-19:
> Wash works works perfectly on OpenWRT 15.05. But we're having issues on OpenWRT-18.06.0 up to 19.07.

what has changed in OpenWRT ? this sounds like some sort of memory corruption happening, maybe due to endianness issues, buggy or too aggressive compiler/compiler settings, buggy libc functions...

adde88, 2019-09-21:
There's quite alot of changes from OpenWRT **15.05** to **19.07**, i'll try to list the most important ones i can find below:

Kernel being bumped from **v3.18.x** to **v4.14.x**
OpenWRT 15.05 was using **uClibc-0.9.33.2**, while OpenWRT 19.07 is using **musl-1.1.23** as the default C-library.
Upgrade from **GCC-4.8-linaro** to **GCC-7.4.0**

I've tried tinkering with some of the compiler flags, such as -fPIC and -fPIE, and also compiling without some of the default flags.
Also tried downgrading to the same **libpcap** version we used on our older firmwares (v1.8.1).
As well as compiling another toolchain with **glibc-2.27** instead of the now default **musl**, but the same problem is happening there.

I noticed this issue while beta-testing the Pineapples latest firmware, where they're trying to follow the latest changes from OpenWRT.
The issue also seems to be present on the default OpenWRT firmware for the pineapple-nano. (Such a firmware doesn't exist for the pineapple-tetra)

https://downloads.openwrt.org/snapshots/targets/ar71xx/generic/openwrt-ar71xx-generic-wifi-pineapple-nano-squashfs-sysupgrade.bin

Today i tried debugging it further with valgrind and gdb, and valgrind reported some issues. 
But i'm not sure if it's related to this issue, or if it's a problem with valgrind.

I'm gonna try compiling it with debugging options again later today, to see if that helps.
If you have any tips for other stuff i can try, please share. :)

VALGRIND (normal scan) https://pastebin.com/zvaXHiMR
VALGRIND (--track-origins=yes) https://pastebin.com/niPF5ffi

rofl0r, 2019-09-21:
the segfault in the last log (last line) certainly looks like something that could give further clues (compile wash with CFLAGS=-g3 and run it with gdb, then look at the backtrace when it crashes).

adde88, 2019-09-23:
Ahh, yeah that was a bit odd. Wash never segfaults when launching it normally, so i'm not sure if that segfault is a problem with valgrind or reaver.

As it's not segfaulting, it's not "easy" to pinpoint the issue with gdb.
Do you have a tip for any certain functions to assign a breakpoint to perhaps? 
I've compiled it with **CFLAGS=-g3**

Also, i tried launching wash today with a few other arguments, and let it run a for a while, and i got this output:
```
root@pinepple-tetra:~# wash -i wlan1mon -2 -5 -a -s -U -F
BSSID               Ch  dBm  WPS  Lck  Vendor    ESSID
--------------------------------------------------------------------------------
53:4F:9E:5E:00:00    6  -49                      (null)
80:2A:A8:1F:3B:9A  153  -85                      (null)
81:67:C4:55:00:00    1  -33                      (null)
52:54:00:14:DA:AA   44  -87                      (null)
52:0D:1F:DF:00:00    3  -59                      (null)
50:69:04:D5:00:00    3  -54                      (null)
52:66:10:86:00:00    3  -54                      (null)
82:A4:80:29:00:00  112  -74                      (null)
51:3B:CB:E3:00:00    3  -78                      (null)
51:90:2E:3C:00:00    3  -76                      (null)
80:86:0A:58:00:00    3  -52                      (null)
81:7F:2C:2A:00:00    3  -75                      (null)
51:3B:15:37:00:00    3  -54                      (null)
51:84:C1:2B:00:00   11  -86                      (null)
82:F6:F2:DE:00:00   11  -85                      (null)
50:82:62:28:00:00    3  -79                      (null)
80:59:76:BC:00:00    3  -82                      (null)
```

As you can see above almost all MAC addresses have the last two octets with only zeros, and checking manually with airodump, or kismet, those MAC addresses are not correct.

If i run wash on a Kali laptop in the same environment i get this correct output (**--all** argument enabled):
```
root@kali:~# wash -i wlan0mon -2 -5 -a
BSSID               Ch  dBm  WPS  Lck  Vendor    ESSID
--------------------------------------------------------------------------------
7A:8A:20:4C:9F:F1    1  -88            AtherosC  Malselvfoto-Guest
8A:8A:20:4C:9F:F1    1  -86            AtherosC  (null)
24:F5:A2:EA:E8:43    3  -58  2.0  No   AtherosC  Lovmo
00:13:37:A5:C2:0F    4  -60                      (null)
B0:B9:8A:69:8C:5A    4  -86  2.0  No   RalinkTe  Sletta12_EXT
02:13:37:A5:C2:0F    4  -61                      TETRA
18:31:BF:E4:26:F8    4  -87  2.0  Yes  Broadcom  Sletta12
38:D5:47:E3:39:A8    6  -48  2.0  No   Broadcom  Nilsen
28:C6:8E:8E:78:48    6  -86  1.0  No   Broadcom  NETGEAR16
08:BD:43:AF:8E:C8    6  -83  2.0  No   AtherosC  NETGEAR74
5C:6A:80:A6:1C:88    9  -90            Broadcom  Sletta6
2C:FD:A1:0E:29:90   13  -71  1.0  No   RalinkTe  Vorhauh2
2C:FD:A1:0E:29:94   36  -85            RalinkTe  Vorhaug2
24:F5:A2:EA:E8:44   36  -60  2.0  No   AtherosC  Lovmo5G
38:D5:47:E3:39:AC  100  -43  2.0  No   Quantenn  Nilsen
78:8A:20:4C:9F:F1    1  -90            AtherosC  Malselvfoto-Adm
40:B0:76:50:46:4C   48  -87            RalinkTe  Kenneth_5GHz
```

rofl0r, 2019-09-23:
yeah, some sort of memory corruption issue. try to compile wash/reaver with address sanitizer (-fsanitize-address etc), then the binary when run should crash properly so gdb can tell what happens (and the sanitizer prints a report too)

rofl0r, 2019-09-23:
btw this reminds me of the driver issue we recently discovered and fixed in https://github.com/t6x/reaver-wps-fork-t6x/issues/279 . so it might make sense to also try to attach some USB device with another chip and see what happens there.

rofl0r, 2019-09-23:
oh actually, the easiest thing to debug this would be a pcap file created via wireshark/tcpdump/airodump-ng, while wash is getting/displaying the trash data.

adde88, 2019-09-25:
Alright, thanks for your suggestions! :)
I'm gonna try compiling another toolchain with support for sanitizer tonight.

I tried your suggestion about using another wifi device. (Both my wn722n and wn821n)
The same issue is occurring there.

I've also uploaded three captures from tcpdump while this issue was happening:
[wash-debug.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/3653559/wash-debug.zip)

Screenshots below, with pcap filenames for context.

## wash-normal-1.pcap
[image]

## wash-F+A-1.pcap
[image]

## wash-normal-tplink-debug.pcap
[image]

rofl0r, 2019-09-25:
thanks, i can't find anything out of the ordinary with the pcaps. if you run wash -f xxx.pcap  on your both devices (embedded / desktop) does it reproduce the bug ?

adde88, 2019-09-26:
Running the captures from tcpdump through wash works perfectly, both on the Pineapple and my desktop.
The issue only seems to affect live captures with wash.

Running reaver on the affected device also works, example: ```reaver -i wlan1mon -b 38:D5:47:E3:39:A8 -c 6 -vvv -K```
Reaver even detects the correct ESSID when attacking the above target: ```Associated with 38:D5:47:E3:39:A8 (ESSID: Nilsen)```

rofl0r, 2019-09-26:
it seems the "live" pcap driver somehow delivers packets that are not on packet boundaries. for example frame 375 in tplink.pcap contains the 4 bytes shown as bssid in the screenshot as the last 4 byte, the 00:00 in turn comes from the first 2 bytes of packet 376. this could be due to all kinds of reasons, so i guess at that point we should look into getting results from the different sanitizers, or if that doesnt work out either, find a way i can debug the issue via ssh or something like that.

adde88, 2019-09-27:
Ahh yeah, i can see the same goes for frame 605. 

I'm working on compiling another toolchain with glibc and ASAN. But sanitizer is disabled by default on the openwrt toolchains, and needs tweaking to get it to compile correctly.

But, i can easily setup SSH access for you on the device.
Just drop me an email on adde88@gmail.com so i can send you the details.

rofl0r, 2019-09-27:
> Just drop me an email

mmh, i don't have email setup currently. it'd be nice if you can show up on irc (irc.freenode.net channel #wash ) to talk about it.

adde88, 2019-10-03:
I'm uploading some pcap files here, generated by my fork of reaver, where i modified wash to create a pcap file of the stuff that it sees internally while running.
I thought it could be relevant to help solving this issue.
Wash was running on the device i'm having issues with.

Some stuff i noticed in the pcap's is that the frame length is always two bytes extra, compared to the exact same frames i get when running my version of wash on my kali desktop.

And those "buggy MAC" addresses i noticed in wash, when looking at the pcap: all those MAC are actually the last 4 bytes in RTS frames (Request-to-send), plus the next two bytes in the next RTS frame. Which i think should actually be the Frame Check Sequence?

Files attached:
[wash-scans.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/3687462/wash-scans.zip)

The zip file contains 3 pcap's, as well as 3 text files that shows the output from wash for that session.

wash-scan-1.pcap: Wash was having issues from the start, left it running for a long time.
wash-scan-2.pcap: Wash worked the first second, then started bugging out.
wash-scan-3.pcap:  Same as above, i just let it run for a longer time. It eventually started displaying Bogus MAC's as well.

rofl0r, 2019-10-05:
i just pushed a couple changes 37de8c8...6f1c14f which make wash -O log every single packet, so we can make better comparisons between packets seen by tcpdump and wash.

adde88, 2019-10-05:
Awesome :)
The bug seems to be gone now. I just compiled it natively on the device, and it seems to work.
I'm gonna try cross-compiling it with the openwrt toolchain, and give a final feedback/close the issue. 👍

rofl0r, 2019-10-05:
it seems the underlying issue is https://bugs.openwrt.org/index.php?do=details&task_id=2197 and we fixed it accidentally in d5e2c820d3b35c6c70b2410571fce293888eec27 by removing the BPF filter.

adde88, 2019-10-06:
Confirmed fixed. Tested both natively and with the normal cross-compiler toolchain :+1:

closed 2019-10-06 by adde88