t6x/reaver-wps-fork-t6x #289: Wash and reaver wont work on Alfa awus036ach

jmaravi, 2019-09-21:
I have my card in monitor mode:
wlan0     IEEE 802.11b  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency:5.745 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=-100 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

However when running wash I got the following error:
root@kali:~# wash -i wlan0
[X] ERROR: pcap_activate status -1
[X] PCAP: generic error code
couldn't get pcap handle, exiting

Same error when trying to run reaver.

The card works fine with airmon, airodump-ng, etc. I was able to capture a 4 way handshake. I am not sure why wash and reaver won't work

jmaravi, 2019-09-22:
Close this issue, I re-installed the latest version of master and now both are working

closed 2019-09-22 by jmaravi

ViliamKovac1223, 2020-01-15:
@jmaravi how did you exactly fix this issue ?

kimocoder, 2020-01-15:
I pushed a fix for this issue to Kali. It Will take 2-3 days before all servers are updated.

ViliamKovac1223, 2020-01-15:
@kimocoder thanks. I will update my system in few days.

blackopscrazy0, 2020-02-12:
I'm still having this issue, any help would be great, thanks.