t6x/reaver-wps-fork-t6x #291: WPS version + lock info vanish after reaver

devocator, 2019-10-20:
After trying reaver with some certain AP's, the wps version and wps lock information vanish and reaver won't work anymore. Airodump shows wps version as 0.0

for e.g.
wash before reaver : B0:4E:26:xx:xx:xx    1  -83  2.0  No   RalinkTe  blablablaSSID
wash after reaver   : B0:4E:26:xx:xx:xx     1  -83                RalinkTe  blablablaSSID

reaver -i wlan0mon -b C4:71:54:32:CD:D8 -c 1 -vv -K

**_REAVER STUCKS AFTER INFO VANISHES_**
```
Reaver v1.6.5 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan0mon to channel 1
[+] Waiting for beacon from B0:4E:26:xx:xx:xx
[+] Received beacon from B0:4E:26:xx:xx:xx
[+] Vendor: RalinkTe

```

rofl0r, 2019-10-20:
yeah, it seems recently some pixiewps-vulnerable devices shut down WPS completely rather than switching to "locked".
it only seems to work on first attempt (so you can actually find the pin with pixiewps), but subsequent attempts cause the router to turn WPS off. i've yet to figure out whether they actually fingerprint and detect reaver specifically (in which case it should work with oneshot.py to get both the pin and submit it later on), or whether they simply reduced the "max wps tries" variable from 3 to 1. in case you have such a model to test with, it would be interesting to make some experiments together. if you're interested, please show up in `#wash` on irc.freenode.net.

devocator, 2019-10-22:
Alright, I'm not that proficient but I'll be there to try my best :)

adde88, 2019-10-24:
Just wanted to chime in on what @rofl0r says about vulnerable devices shutting down WPS on subsequent attempts. I've also experienced this several times now. All of the devies i've encountered has been ASUS AP's.
I could create a list of every device i find, with the exact device name, if this is something that's useful.
I'm a bit curious to exactly what's happening.

soxrok2212, 2019-10-24:
@addee88 it’s been happening for me since around 2017, IIRC many or all were Ralink. Can you confirm the chipsets?

rofl0r, 2019-10-24:
> I'm a bit curious to exactly what's happening.

according to the tests done by @devocator it looks as if the router detects the pixiewps attack of reaver because we interrupt the connection at M3. with https://github.com/rofl0r/oneshot used in pixiewps mode, the router didn't "lock" down in this fashion.
a fix is in the works.

feitoi, 2019-10-24:
I guess the router does not detects the pixiewps attack, only the logic for WPS authentication has been broken, and it to turn WPS off.
When **reaver** interrupts after received M2 and called pixiewps, the router continues to wait for M3 and it stays in "WPS-AUTH" status (using WiFi Analyzer (open-souce) of Android for scan), after timeout (1 minute), router to turn WPS off.
Solutions:
1) send M3 and interrupt attack with (exit ()), this way decreases WPS lock count.
2) send a WSC NACK and interrupt the attack, this way avoid WPS lock.

I tested both ways with a router of mine and it worked. Second form is better because it avoid WPS lock. I don't know will work with other routers. Needs tests.

There is this implementation in my pull request #273

rofl0r, 2019-10-25:
good analysis @feitoi. it's probably indeed best to just send nack, and not continuing further with the protocol. however i already had a PR prepared, see above.
it would be nice if this could be tested.

feitoi, 2019-10-25:
I'm sorry, there are errors in my previous comment.
Where M2 means M3 and M3 means M4.

commit eec6a897 (2019-10-30): pixiewps mode: implement automatic cracking and proper NACK usage

a couple routers started turning WPS off after a pixie attack.
research showed this happens only with reaver, not oneshot, so it's
either a defensive mechanism against reaver itself, or simply against
non-compliant WPS implementations.
as @feitoi pointed out, the best way to deal with that is to send
WSC NACK.

however, we can just as well continue the session if pixiewps succeeded
in a certain timeframe, and inject the found pin into the running
WPS transaction.

the new logic now works as follows:
pixiewps is started in a thr [... cut]
closed 2019-10-30 by rofl0r