t6x/reaver-wps-fork-t6x #293: -Wno-unused-but-set-variable

d33p3r, 2019-10-26:
Hello.

I use BackTrack5 R3. I have a problem with installing reaver-wps-fork-1.6.4:

root@premium:~/reaver-wps-fork-1.6.4/src# ./configure
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for pcap_open_live in -lpcap... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for stdlib.h... (cached) yes
checking for stdint.h... (cached) yes
checking for string.h... (cached) yes
checking pcap.h usability... yes
checking pcap.h presence... yes
checking for pcap.h... yes
checking to see if compiler understands -Werror-unknown-warning-option... no
checking to see if compiler understands -Wno-unused-but-set-variable... yes
configure: creating ./config.status
config.status: creating config.mak

root@premium:/reaver-wps-fork-1.6.4/src# make
cp lwe/wireless.21.h lwe/wireless.h
gcc -DCONF_DIR='"/usr/local/var/lib/reaver"' -Wall -g -O2
-Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-function
-Wno-pointer-sign -Ilibwps -I. -Ilwe -DCONFIG_IPV6 -c -o
argsparser.o argsparser.c
In file included from globule.h:37, from argsparser.c:37:
defs.h:326: error: redefinition of typedef ‘le16’
utils/common.h:428: note: previous declaration of ‘le16’ was here
defs.h:327: error: redefinition of typedef ‘le32’
utils/common.h:430: note: previous declaration of ‘le32’ was here
defs.h:329: error: redefinition of typedef ‘be16’
utils/common.h:427: note: previous declaration of ‘be16’ was here
defs.h:330: error: redefinition of typedef ‘be32’
utils/common.h:429: note: previous declaration of ‘be32’ was here
In file included from argsparser.c:41:
pixie.h:5:1: warning: "_POSIX_SOURCE" redefined
In file included from /usr/include/stdio.h:28, from argsparser.c:34:
/usr/include/features.h:202:1: warning: this is the location of the previous definition
cc1: warning: unrecognized command line option "-Wno-unused-but-set-variable"
make: *** [argsparser.o] Error 1
root@premium:/reaver-wps-fork-1.6.4/src#

The same problem is with 1.6.5 version. All previous versions from 1.4 to 1.6.3 hasn't that error.

Pls help. I don't know, what to do with it...
D33p3r

rofl0r, 2019-10-27:
what gcc version ? 
this repeated typedef of le16 has nothing to do with -Wno-unused-but-set-variable , and it causes no problem with GCC 6.5.0.

d33p3r, 2019-10-27:
root@premium:~# gcc --version
gcc (Ubuntu 4.4.3-4ubuntu5.1) 4.4.3

closed 2019-10-29 by rofl0r

rofl0r, 2019-10-29:
in case you need to build the release 1.6.4, you need to download  https://github.com/t6x/reaver-wps-fork-t6x/commit/01432621e4be6046d50af0d03f4c568ba955ddf7.patch and apply it to the extracted reaver tarball (`patch -p1 < patchname`)

d33p3r, 2019-10-29:
OK, now version 1.6.5 works fine. Thank you! :)

commit 01432621 (2019-10-29): fix compatibility with GCC 4.4

that specific compiler chokes on duplicate typedefs, while newer ones
do not.

closes #293