t6x/reaver-wps-fork-t6x #295: How uninstall/remove/purge this version?

Veigalex, 2019-12-28:
Guys, this version doesn't have "make unistall" and apt remove or purge doesn't remove it.  
When using apt I get the information that the retriever has already been removed, but I type retrieve and the program is called.

Do I have to delete the folders in the / usr / local / bin directory?
Help me!

kcdtv, 2019-12-28:
apt remove the programs that you install with apt, learn how works your system and visit genralist websites for that, github is not for this kind of questions... And yes, simply delete the program wherever it is and the folder reaver to "uninstall it"
use command find if you are lost 
`find / -iname  "reaver"`

closed 2019-12-28 by kcdtv

Veigalex, 2019-12-28:
Thanks!
The programmer could have created make unistall.  :)

kcdtv, 2019-12-30:
The programmer from the first version (craig heffner) did implement `make distclean` but it is not the case in the modern version. You can always open an issue tagged as "enhancement/feature request"  if you want, maybe r0flor or someone else have some time and can do it. Cheers and happy new year.