t6x/reaver-wps-fork-t6x #296: ERROR: pcap_activate status -1, CAP: generic error code

Voyotv, 2020-01-04:
Hello

If I try to run the latest in-built version of Wash/Reaver I get this error:

[image]

So I downgraded to v1.6.1 which runs o.k., but than I get new problems. If I run Reaver on it's own, it just won't associate:

[image]

If I do the association manually than I get this:

[image]

Wifi chip that I'm using is Realtek RTL8812AU. Injection and everything else seems to work just fine. Anyone knows what could be the problem here?

kimocoder, 2020-01-11:
Download reaver from source, cd "reaver-wps-fork-t6x/src" and "make && make install"

@rofl0r could you tag a new version?

Error0x00000001, 2020-02-06:
kimokoder fork Work! ))

kimocoder, 2020-02-06:
Great

jarilaos, 2020-06-03:
Hi!

Yesterday I updated Reaver from 1.6.5 to 1.6.6 and the issue is fixed. 

I'm using Kali with kernel 5.6 and the latest Realtek RTL8812AU chipset 5.6.4.2~20200529-0kali1.

You could close the issue if you want.

Thanks for all contributions!

closed 2020-06-04 by soxrok2212