t6x/reaver-wps-fork-t6x #298: Failed to obtain WPS pin on my ISP router but someone managed to get my wireless password via reaver?

hexclann, 2020-01-10:
I ran a Pixiedust attack after i came to know that i was compromised by someone, I have a secure password but i cannot disable wps on my ISP router because it is locked down. To test it myself, i ran 

reaver -i wlan0mon -b -c -K -N -vvv but it says failed to get WPS pin, here are the outputs:

https://www.pastiebin.com/5e18692467e59

I want to verify if my router is vulnerable or not before purchasing a new one

soxrok2212, 2020-01-10:
Perhaps your password isn’t as secure as you think it is, or maybe someone else you live with fell for a FakeAP. 

If the latest Reaver doesn’t crack it and you’re positive they broke it with Reaver, then someone knows something we don’t.

closed 2020-01-10 by soxrok2212

hexclann, 2020-01-10:
@soxrok2212 I think someone fell for a FakeAP because the router seems to be protected from WPS attack