t6x/reaver-wps-fork-t6x #300: A problem with make

CWSHBR, 2020-01-22:
clang-9: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:123: reaver] Error 1

adde88, 2020-01-22:
It's compiling perfectly fine with clang for me.

CWSHBR, 2020-01-23:
gcc -DCONF_DIR='"/usr/local/var/lib/reaver"' -Wall -g -O2 -Werror-unknown-warning-option -Wno-unused-variable -Wno-unused-function -Wno-pointer-sign -Ilibwps -I. -Ilwe -DCONFIG_IPV6 -Ilibwps -I. -Ilwe globule.o init.o sigint.o iface.o sigalrm.o misc.o session.o send.o pins.o 80211.o builder.o keys.o crc.o pixie.o version.o pcapfile.o exchange.o argsparser.o wpscrack.o wpsmon.o cracker.o main.o libwps/libwps.o wps/wps_attr_build.o wps/wps_attr_parse.o wps/wps_attr_process.o wps/wps.o wps/wps_common.o wps/wps_dev_attr.o wps/wps_enrollee.o wps/wps_registrar.o wps/wps_ufd.o utils/base64.o utils/common.o utils/ip_addr.o utils/radiotap.o utils/trace.o utils/uuid.o utils/wpa_debug.o utils/wpabuf.o utils/os_unix.o utils/vendor.o utils/eloop.o tls/asn1.o tls/bignum.o tls/pkcs1.o tls/pkcs5.o tls/pkcs8.o tls/rsa.o tls/tlsv1_client.o tls/tlsv1_client_read.o tls/tlsv1_client_write.o tls/tlsv1_common.o tls/tlsv1_cred.o tls/tlsv1_record.o tls/tlsv1_server.o tls/tlsv1_server_read.o tls/tlsv1_server_write.o tls/x509v3.o crypto/aes-cbc.o crypto/aes-ctr.o crypto/aes-eax.o crypto/aes-encblock.o crypto/aes-internal.o crypto/aes-internal-dec.o crypto/aes-internal-enc.o crypto/aes-omac1.o crypto/aes-unwrap.o crypto/aes-wrap.o crypto/des-internal.o crypto/dh_group5.o crypto/dh_groups.o crypto/md4-internal.o crypto/md5.o crypto/md5-internal.o crypto/milenage.o crypto/ms_funcs.o crypto/rc4.o crypto/sha1.o crypto/sha1-internal.o crypto/sha1-pbkdf2.o crypto/sha1-tlsprf.o crypto/sha1-tprf.o crypto/sha256.o crypto/sha256-internal.o crypto/crypto_internal.o crypto/crypto_internal-cipher.o crypto/crypto_internal-modexp.o crypto/crypto_internal-rsa.o crypto/tls_internal.o crypto/fips_prf_internal.o lwe/iwlib.o -lm -lpcap  -lpthread -o reaver
/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: wpsmon.o: in function `monitor':
/data/data/com.termux/files/home/reaver-wps-fork-t6x/src/wpsmon.c:332: undefined reference to `ualarm'
/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: wpsmon.o: in function `parse_wps_settings':
/data/data/com.termux/files/home/reaver-wps-fork-t6x/src/wpsmon.c:433: undefined reference to `ualarm'
clang-9: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:123: reaver] Error 1

rofl0r, 2020-01-23:
> aarch64-linux-android-ld: wpsmon.o: in function monitor': /data/data/com.termux/files/home/reaver-wps-fork-t6x/src/wpsmon.c:332: undefined reference to ualarm'

android isn't supported.

closed 2020-01-23 by rofl0r