t6x/reaver-wps-fork-t6x #301: [X] ERROR: pcap_activate status -1

SoulSA, 2020-01-23:
Hello,

I have a Kali Linux machine, and an Alfa AWUS036ACH wireless adapter running on "RTL8812AU"
When I try "wash --interface wlan1" command, I get the error:
<code>
[X] ERROR: pcap_activate status -1
[X] PCAP: generic error code
couldn't get pcap handle, exiting</code>

I trid the solution that @rofl0r mentioned in the <a href="https://github.com/t6x/reaver-wps-fork-t6x/pull/276">276</a>, but it didn't work for me.

I tried the reaver command, and it game me the same error:

<code>
Reaver v1.6.5 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[X] ERROR: pcap_activate status -1
[X] PCAP: generic error code
couldn't get pcap handle, exiting
</code>

Please help me solve this issue.

Operating System: Kali GNU/Linux Rolling
Kernel: Linux 5.4.0-kali2-amd64

adde88, 2020-01-23:
Is wlan1 in monitor mode? Does injection work?
Try testing the device with: aireplay-ng -9 wlan1

SoulSA, 2020-01-23:
Yes, it is in monitor mode, and the injection works fine.
The problem started when I was doing the "wash & reaver" commands, everything I tried did not work.

kimocoder, 2020-01-23:
I've updated the drivers some days ago to fix this issue, it's possible you need to clone the repo and build it. You may also clone Reaver and build it as the latest isn't pushed to Kali.

SoulSA, 2020-01-23:
@kimocoder : I unistalled the old driver and I cloned the one from your repos <a href="https://github.com/kimocoder/rtl8812au">rtl8812au</a> and followed the instructions, rebooted the machine, and the adapter did not show in the wireless tap. Tried to install it again, and I got this:

<pre><code>root@kali:~/rtl8812au# ./dkms-install.sh 
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.6.4.2/source ->
                 /usr/src/rtl8812au-5.6.4.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j8 KVER=5.4.0-kali2-amd64 KSRC=/lib/modules/5.4.0-kali2-amd64/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.0-kali2-amd64 (x86_64)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j8 KVER=5.4.0-kali2-amd64 KSRC=/lib/modules/5.4.0-kali2-amd64/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.0-kali2-amd64 (x86_64)
Consult /var/lib/dkms/rtl8812au/5.6.4.2/build/make.log for more information.
Finished running dkms install steps.</code></pre>

What is the problem?

kimocoder, 2020-01-23:
Oooh virtualbox.. these drivers don't really like virtual environments, as the README says. Hard to say whats up with your setup and DKMS, but try building without DKMS. Open a issue report over at us instead of us spamming Reaver issue reports

SoulSA, 2020-01-23:
I will close this one, and I will give it another go after removing virtual-box from my machine. I will report to you.

@kimocoder @adde88 Thank you both for your help.

closed 2020-01-23 by SoulSA