t6x/reaver-wps-fork-t6x #302: RTL8812AU (AWUS036ACH) 5.6.4.2 Timeout

Error0x00000001, 2020-02-01:
root@book:/home/igor/rtl8812au# reaver 
Reaver v1.6.5-git-70-g1c86168 WiFi Protected Setup Attack Tool
######################
5.6.4.2 - Not Worked 
5.2.20.2  Worked
######################
[+] Trying pin "12345670"
[+] Sending authentication request
[+] Sending association request
[+] Associated with F8:BF:09:8A:E4:10 (ESSID: MgtsGpon83)
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received deauth request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received deauth request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[!] WARNING: 10 successive start failures
[+] Sending EAPOL START request
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin "12345670"

soxrok2212, 2020-02-01:
*sigh* Perhaps we should add an issue template... can you give us any information besides it doesn’t work?

Error0x00000001, 2020-02-01:
Kali, image 5.2*, 5.3*,5.4*
Wash worked
Reaver Not worked (Timeout only)

rofl0r, 2020-02-01:
https://github.com/t6x/reaver-wps-fork-t6x/wiki/Troubleshooting

Error0x00000001, 2020-02-06:
Makr clean - Fine!
Work!

closed 2020-02-06 by Error0x00000001