t6x/reaver-wps-fork-t6x #306: Sending EAPOL START request timeout

starraiderx, 2020-02-16:
Hello I am pretty new to kali Linux and followed a guide from Nullbyte for a pixie dust attack.
I tested with aireplay-ng --test wlan0mon my Intel 7265D if it supports Injection and it says it does. 
The following is the console output on kali linux everything should be updated and all the dependencies from Wifite2 are installed 

sudo reaver -i wlan0mon -b 78:DD:12:C4:A3:80 -c 11 -vvv -K 1 -f

Reaver v1.6.5-git-69-geebc0e9 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan0mon to channel 11
[+] Waiting for beacon from 78:DD:12:C4:A3:80
[+] Received beacon from 78:DD:12:C4:A3:80
[+] Vendor: Unknown 
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
     31 32 33 34 35 36 37 30                           12345670        
WPS: Selected registrar information changed
WPS: Internal Registrar selected (pbc=0)
WPS: sel_reg_union
WPS: set_ie
WPS: cb_set_sel_reg
WPS: Enter wps_cg_set_sel_reg
WPS: Leave wps_cg_set_sel_reg early
WPS: return from wps_selected_registrar_changed
[+] Trying pin "12345670"
send_packet called from deauthenticate() 80211.c:380
send_packet called from authenticate() 80211.c:411
[+] Sending authentication request
send_packet called from associate() 80211.c:464
[+] Sending association request
[+] Associated with 78:DD:12:C4:A3:80 (ESSID: Anonym)
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[!] Found packet with bad FCS, skipping...
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request

soxrok2212, 2020-03-06:
Intel cards are notoriously bad for injection/monitor mode support. Please try with a known working chipset/wireless card (RTL8187, AR9271, etc).

closed 2020-03-06 by soxrok2212

Charicharan, 2020-03-17:
I'm using rtl8188eu chipset and I'm too getting the same kind of timeout and errors is this a chipset problem??

rofl0r, 2020-08-24:
according to your screenie injection works, which is a good start. it's odd though that you get the ioctl warning, do you have wpa_supplicant or network manager running? if so these need to be disabled.
also check the "troubleshooting guide" on the wiki.

Tanticion, 2021-03-21:
This is actually a bug with reaver. try downgrading the version of reaver.

rofl0r, 2021-03-21:
> This is actually a bug with reaver. try downgrading the version of reaver.

oh? which version of reaver would fix it, in your opinion ?

Tanticion, 2021-03-27:
Try 1.6.1

rofl0r, 2021-03-27:
if there's an actual regression, i'd appreciate if you could `git bisect` the commit that introduced it.

udaykiran192, 2022-03-02:
Hey, I tried downgrading reaver-1.6.6 to reaver-1.6.1 but I got error when I type make 
[image]
What I did is
# apt- get update
# apt-get upgrade 
downloaded reaver-1.6.1 from github and extracted the file 
# cd reaver-1.6.1
# cd src
# . /configer 
# make 
I got this error 
[image]
 
Please help me out of this iam a biggener to kali