t6x/reaver-wps-fork-t6x #307: New release?

nckx, 2020-02-24:
Hi,

Thanks for your work on this!  Would you be willing to cut a new release from current master?

I noticed that [some bugfixes from 2018](https://github.com/t6x/reaver-wps-fork-t6x/compare/v1.6.5...master) haven't seen a release yet, and would like to include them in [my operating system's](https://guix.gnu.org/) package.

soxrok2212, 2020-03-06:
CC @rofl0r @kcdtv.Haven't seen either of them on IRC in some time...

kcdtv, 2020-03-09:
Hi there! 
I hope you are doing fine.
   Let's do it! 
Not only because it has been a long time since last release but because there are some really important fixes. 
I am thinking for example about all the work made about sessions. 
It deserves visibility with a release note and a little wiki article (I can do that). 
@ nckx: Congrats for all your work, very nice!

kimocoder, 2020-03-16:
New release added by @rofl0r 
This issue may be closed.

rofl0r, 2020-03-16:
https://github.com/t6x/reaver-wps-fork-t6x/releases/tag/v1.6.6

closed 2020-03-16 by rofl0r

rofl0r, 2020-03-16:
thanks also to @soxrok2212 who helped with the release

rofl0r, 2020-03-17:
> I am thinking for example about all the work made about sessions.
> It deserves visibility with a release note and a little wiki article (I can do that).

@kcdtv i just saw this paragraph, it would be really cool if you could supply that wiki article, thanks! also feel free to edit the release notes if you feel something important is missing.

kcdtv, 2020-03-27:
HI! I did not see that you published the new release (the 2 lasts weeks have been very weird here...), well done guys!

>  it would be really cool if you could supply that wiki article

I guess i will be able to find some time in the next weeks :stuck_out_tongue_closed_eyes:  Will do!