t6x/reaver-wps-fork-t6x #308: Problem with RT3070 and RTL8192CU

vincenzogianfelice, 2020-02-24:
Hi. I tried both to carry out the brute force and to insert the correct wps pin manually ... but there was nothing to do.
It always brings me this:

```
Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Sending WSC NACK
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
```
If I try with another chipset (like the very famous AWUS036NH)
everything works perfectly ...
The **rt3070** and **rtl8192cu** chipsets work very well with aircrack-ng (I did a test with airepleay) and it's all right (the only problem, it would seem to be the **attack -5** (**fragmentation**))

soxrok2212, 2020-03-06:
Have you tried adding -N (no nacks) in your attack parameters?

kcdtv, 2020-03-09:
This message is not clear at all.
If it works with **AWUS036NH** then it works with **RT3070** because that it is its chipset,
Please follow the procedure for debugging: [Troubleshooting](https://github.com/t6x/reaver-wps-fork-t6x/wiki/Troubleshooting) 
If there is an issue we need full information with complete stdout with full verbose.

rofl0r, 2020-03-16:
RT3070 indeed requires -N parameter. please reopen this if you still have issues after adding that.

closed 2020-03-16 by rofl0r