t6x/reaver-wps-fork-t6x #310: Reaver not working on my AP with WPS2.0

fwag, 2020-03-25:
Hi, 
it seems that reaver is not working on WPS 2.0 of my router.
It always fails and retry the same password.
At the beginning I thought the AP was not supporting the PIN, but looking at one ProbeResponse coming from the AP (Source Mac Address = one of the AP) I see the following:

[image]

where I see that KEYPAD (PIN) is supported.

Are there any constraints with WPS 2.0 ?

Regards
Fwag

RLlane2126, 2020-03-26:
try a new api

fwag, 2020-03-26:
Which new api ?
This is the pcap collected after running reaver on my AP.

AP: 00:00:00:11:11:11
Client (Reaver): 00:00:00:22:22:22

Do you know why is not moving forward and gets stuck in request/response identity?

[test.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/4386918/test.zip)

kcdtv, 2020-03-27:
The fact that the mode is supported does not mean that it is properly enabled...
Be sure that your have a PIN correctly defined in your router web interface and that the protocol is fully enabled
If you still have an issue after that follow this steps: [Troubleshooting](https://github.com/t6x/reaver-wps-fork-t6x/wiki/Troubleshooting) 

> Are there any constraints with WPS 2.0 ?

Router lock after 10 attempts became "mandatory"  but that does not change anything in the way to associate and send a PIN.

closed 2021-04-18 by kcdtv