t6x/reaver-wps-fork-t6x #311: error unknown type name 'pcap_t'

kos-crypto, 2020-04-07:
root@:~/reaver-wps-fork-t6x/src# make
gcc -DCONF_DIR='"/usr/local/var/lib/reaver"' -Wall -g -O2 -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-function -Wno-pointer-sign -Ilibwps -I. -Ilwe -DCONFIG_IPV6   -c -o globule.o globule.c
In file included from globule.c:34:
globule.h:149:9: error: unknown type name ‘pcap_t’
  149 |         pcap_t *handle;                 /* Pcap handle */
      |         ^~~~~~
globule.h:251:17: error: unknown type name ‘pcap_t’
  251 | void set_handle(pcap_t *value);
      |                 ^~~~~~
globule.h:252:1: error: unknown type name ‘pcap_t’
  252 | pcap_t *get_handle();
      | ^~~~~~
globule.c: In function ‘globule_deinit’:
globule.c:70:23: warning: implicit declaration of function ‘pcap_close’; did you mean ‘pclose’? [-Wimplicit-function-declaration]
   70 |   if(globule->handle) pcap_close(globule->handle);
      |                       ^~~~~~~~~~
      |                       pclose
globule.c: At top level:
globule.c:529:17: error: unknown type name ‘pcap_t’
  529 | void set_handle(pcap_t *value)
      |                 ^~~~~~
globule.c:533:1: error: unknown type name ‘pcap_t’
  533 | pcap_t *get_handle()
      | ^~~~~~
make: *** [<builtin>: globule.o] Error 1

rofl0r, 2020-04-08:
do you have libpcap-dev installed ?

kos-crypto, 2020-04-09:
[image]

feitoi, 2020-04-22:
Setups again your environment
```BASH
make clean
./configure
make
```

kos-crypto, 2020-04-22:
> Setups again your environment
> 
> ```shell
> make clean
> ./configure
> make
> ```

[image]
[image]
[image]