t6x/reaver-wps-fork-t6x #318: Error : Pcap_activate status -1

Alamnzr123, 2020-05-07:
`ERROR : pcap_activate status -1
PCAP : generic error code
Couldn't get pcap handle, exiting`

Hello, i am getting this message when i type " wash -i <interface>"
I already install new version and still getting this issue.

rofl0r, 2020-07-09:
you didn't specify a network interface ?

jesse-s07, 2021-07-29:
Estou tendo esse mesmo problema citado acima, alguém poderia mim ajudar a resolver:

ERROR : pcap_activate status -1 PCAP 
generic error code Couldn't get pcap handle, exiting 

essa é a mensagem de erro que recebo quando executo o reaver: "reaver -i wlan0mon -b 00:00:00:00:00 -v"

Aftondil, 2023-05-15:
Hello I have problem 
[X] ERROR: pcap_activate status -8 
When I type reaver -i (interface) -b 00000 -vv
Anyone knows how to solve?