t6x/reaver-wps-fork-t6x #319: Error Code 0x02

krypto-kraken, 2020-05-21:
Hey guys,

Not sure what this error code means...but I continually get it when trying different APs that I have enabled WPS on. The one from the output below is running WPS 1.0. I've tried letting Reaver associate, and I have tried using aireplay to do the association. Any insight into what the potential problem could be would be great. I am using an Alfa AWUS036H. I also have the Wireshark dumps available if needed but I couldn't glean any real issues from that either. Just the AP eventually sending an EAPOL Failure. Thanks!!

`Reaver v1.6.6 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan0 to channel 6
[+] Waiting for beacon from C0:3F:0E:06:C8:78
[+] Received beacon from C0:3F:0E:06:C8:78
[+] Vendor: Broadcom
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
     31 32 33 34 35 36 37 30                           12345670        
WPS: Selected registrar information changed
WPS: Internal Registrar selected (pbc=0)
WPS: sel_reg_union
WPS: set_ie
WPS: cb_set_sel_reg
WPS: Enter wps_cg_set_sel_reg
WPS: Leave wps_cg_set_sel_reg early
WPS: return from wps_selected_registrar_changed
[+] Trying pin "12345670"
[+] Associated with C0:3F:0E:06:C8:78 (ESSID: NETGEAR01)
[+] Sending EAPOL START request
send_packet called from send_eapol_start() send.c:48
[+] Received deauth request
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
WPS: Processing received message (len=392 op_code=4)
WPS: Received WSC_MSG
[... 709 lines cut ...]
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
[+] Received identity request
[+] Sending identity response
send_packet called from send_identity_response() send.c:81
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
send_packet called from resend_last_packet() send.c:161
^Csend_packet called from send_termination() send.c:142
`

rofl0r, 2020-07-07:
did you read the troubleshooting guide in the wiki ?

krypto-kraken, 2020-07-23:
Yeah I did. Did I miss something I should have figured out myself (sorry if I did)?

The AP requests the identity of the client, and the client responds, then does an extended response when it does not see an answer from the AP. Then all of a sudden the AP sends a failure. I know from Stefan Viehböck's paper that we tell the AP that we are attempting to become the registrar (which I believe to be different than how a normal client would authenticate with WPS - they'd become the enrollee correct?). Does the fact that the AP responds with a failure right after we state we are the registrar mean that the AP does not support this method (and thus the attack)? I ensure a few times that the WPS PIN method was enabled on the AP.

Thanks for your help! Just trying to figure out when I can discount WPS as a valid attack method, and when I should pursue it. 

[image]

rofl0r, 2020-07-24:
reaver sends a NACK because it runs into the configured timeout. i guess you should try to set a higher timeout with the appropriate command line option